project-chip/connectedhomeip #21238: Address Android handling of short discriminators

bzbarsky-apple, 2022-07-26:
#### Problem
In https://github.com/project-chip/connectedhomeip/pull/21187 I left the Android behavior around short discriminators the same as it was before: they are 0-padded to look like a long discriminator with all low bits 0. This happens in two places:

1. BLEManagerImpl::NewConnection in src/platform/android/BLEManagerImpl.cpp
2. TransformSetupPayload in src/setup_payload/java/SetupPayloadParser-JNI.cpp

 #### Proposed Solution
What needs to happen is to figure out how short discriminators should actually be handled in the Android API and handle them accordingly.

123yangu, 2023-01-03:
@bzbarsky-apple Do I have this problem to solve?

bzbarsky-apple, 2023-01-03:
@123yangu I don't understand what you're asking.

yunhanw-google, 2023-03-11:
@joonhaengHeo  could you hep check whether this has been resolved. thanks

joonhaengHeo, 2023-07-20:
I already added pairWithCode API. (https://github.com/project-chip/connectedhomeip/pull/26324)
And, if using Manual Pairing Code, proceed with BLE discover with Short Discriminator.

closed 2023-07-20 by joonhaengHeo