project-chip/connectedhomeip #21255: build Failed

yinhejianke, 2022-07-27:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

ninja: Entering directory `./out/debug'
[1733/3907] c++ linux_x64_gcc/obj/src/lib/support/tests/libSupportTests.TestStateMachine.cpp.o
FAILED: linux_x64_gcc/obj/src/lib/support/tests/libSupportTests.TestStateMachine.cpp.o
g++ -MMD -MF linux_x64_gcc/obj/src/lib/support/tests/libSupportTests.TestStateMachine.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-unknown-warning-option -Wno-maybe-uninitialized -Wno-missing-field-initializers -Wno-unused-parameter -Wno-psabi -Wno-cast-function-type -fdiagnostics-color -fno-strict-aliasing -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-4 -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -Wno-implicit-fallthrough -Wno-implicit-fallthrough -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Ilinux_x64_gcc/gen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -I../../third_party/inipp/repo/inipp -I../../third_party/nlunit-test/repo/src -I../../third_party/jsoncpp/repo/include -c ../../src/lib/support/tests/TestStateMachine.cpp -o linux_x64_gcc/obj/src/lib/support/tests/libSupportTests.TestStateMachine.cpp.o
../../src/lib/support/tests/TestStateMachine.cpp: In constructor ‘{anonymous}::SimpleStateMachine::SimpleStateMachine()’:
../../src/lib/support/tests/TestStateMachine.cpp:158:41: error: member ‘{anonymous}::SimpleStateMachine::mStateMachine’ is used uninitialized [-Werror=uninitialized]
  158 |     SimpleStateMachine() : mTransitions(mStateMachine), mStateMachine(mTransitions) {}
      |                                         ^~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[1746/3907] c++ linux_x64_gcc/obj/examples/chip-tool/chip-tool.main.cpp.o
ninja: build stopped: subcommand failed.

 #### Proposed Solution
<suggested fix, suggested enhancement>

yinhejianke, 2022-07-27:
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/12.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 12.1.0 (GCC)

yinhejianke, 2022-07-27:
DISTRIB_ID="Arch"
DISTRIB_RELEASE="rolling"
DISTRIB_DESCRIPTION="Arch Linux"
NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
LOGO=archlinux-logo

yinhejianke, 2022-07-27:
[image]

bzbarsky-apple, 2022-07-27:
@msandstedt

turon, 2022-09-07:
I'm seeing this after `git reset --hard origin/master` this morning.  I was up-to-date with master as recently as yesterday.

andy31415, 2022-09-08:
When fixing this, please undo the change in https://github.com/project-chip/connectedhomeip/pull/22461 (the uninitialized should remain an error)

andy31415, 2022-09-08:
Re-opening to fix the circular dependency rather than disabling the warning in the compiler.

commit 660786b2 (2022-09-08): [build] Fix #21255 - allow circular initialization of SimpleStateMachine test. (#22461)

* [build] Fix #21255 - allow circular initialization of SimpleStateMachine test.

* [build] Add comment per review feedback.
commit 96c93578 (2022-10-20): [SVE2] Cherry Pick Matter Official CDs and CD Trusted Storage Implementation  (#23239)

* Add missing pthread header (#22833)

* [build] Fix #21255 - allow circular initialization of SimpleStateMachine test. (#22461)

* [build] Fix #21255 - allow circular initialization of SimpleStateMachine test.

* [build] Add comment per review feedback.

* [Darwin][AttestationVerifier] Expose a mechanism to customise cd signing keys and use it in darwin (#22338)

* Add AttestationTrustStore::GetCertificationDeclarationCert virtual method to allow controllers passing in some CD certs

* Add cdCerts member t [... cut]
closed 2023-03-23 by stale[bot]