project-chip/connectedhomeip #21264: chip-tool end device commissioning ble-thread pairing timeout error

GSIVANAGARJUNA, 2022-07-27:
sudo out/chip-tool pairing ble-thread 102 hex:0e080000000000010000000300000f35060004001fffe0020811111111222222220708fd39728522d0950a051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f01021234041061e1206d2c2b46e079eb775f41fc72190c0402a0fff8 20202021 3840

while using the above chip-tool command I am getting the below error please resolve the issue

CHIP:TOO:Run command failure  ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:244:CHIP Error ox00000032:timeout

bzbarsky-apple, 2022-07-27:
@GSIVANAGARJUNA Please attach (not paste) complete logs from chip-tool and the server side of this interaction.

GSIVANAGARJUNA, 2022-07-28:
@bzbarsky-apple please find attached log files 
 
[Timeout_error.txt](https://github.com/project-chip/connectedhomeip/files/9206278/Timeout_error.txt)
[matter lo terminal.txt](https://github.com/project-chip/connectedhomeip/files/9206280/matter.lo.terminal.txt)

closed 2022-07-28 by GSIVANAGARJUNA

bzbarsky-apple, 2022-07-28:
@GSIVANAGARJUNA Looks like operational discovery failed. Did the server actually join the Thread network?  Were its operational advertisements visible?

GSIVANAGARJUNA, 2022-08-03:
in matter lighting example end device terminal it shows following error
<error > [DIS] Failed to advertise extended commissionable node: 3

Jakebluhm, 2022-10-14:
@GSIVANAGARJUNA Did you end up figuring this issue out? I'm having the same problem commissioning a nRF52840 with the chip-tool running on Mac (MacOS Monterey 12.6)