project-chip/connectedhomeip #21277: [SVE] TC-CNET-4.11 need clarity about the working of chip-tool side

chirag-silabs, 2022-07-27:
#### Problem
TC-CNET-4.11 networkcommissioning wifi connect-networks
1. The DUT must be disconnected to AP1 and connect to AP2(changes done on sdk)
2. The step 7 where DUT is connected to the second access point, we are not getting ConnectNetworkResponse since the network is changed and the chip-tool running on pi is on 1st AP. The DUT is connected to AP2 but chip-tool on raspberrypi is failing with timeout.
3. After disarming the failsafe the DUT is not getting connected to the AP.
 * expected behavior
 step 7:
  [1653479953.633815][30491:30496] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0007
  [1653479953.633986][30491:30496] CHIP:TOO:   ConnectNetworkResponse: {
  [1653479953.634133][30491:30496] CHIP:TOO:     networkingStatus: 0
  [1653479953.634195][30491:30496] CHIP:TOO:     errorValue: null
  [1653479953.634275][30491:30496] CHIP:TOO:    }
  [1653479953.634367][30491:30496] CHIP:DMG: ICR moving to [AwaitingDe]
  [1653479953.634501][30491:30496] CHIP:EM: Sending Standalone Ack for MessageCounter:490923 on exchange 21425i
 * actual behavior
  [1658929461.640140][2633:2633] CHIP:CTL: Shutting down the commissioner
  [1658929461.640193][2633:2633] CHIP:CTL: Stopping commissioning discovery over DNS-SD
  [1658929461.640216][2633:2633] CHIP:CTL: Shutting down the controller
  [1658929461.640252][2633:2633] CHIP:IN: Expiring all sessions for fabric 0x1!!
  [1658929461.640277][2633:2633] CHIP:IN: SecureSession[0xffff88001930]: MarkForEviction Type:2 LSID:5697
  [1658929461.640300][2633:2633] CHIP:SC: SecureSession[0xffff88001930]: Moving from state 'kActive' --> 'kPendingEviction'
  [1658929461.640346][2633:2633] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 48555i
  [1658929461.640383][2633:2633] CHIP:TOO: Error: ../../src/app/CommandSender.cpp:214: CHIP Error 0x00000032: Timeout
  [1658929461.640405][2633:2633] CHIP:DMG: ICR moving to [AwaitingDe]
  [1658929461.640473][2633:2633] CHIP:IN: SecureSession[0xffff88001930]: Released - Type:2 LSID:5697
  [1658929461.640504][2633:2633] CHIP:FP: Forgetting fabric 0x1
  [1658929461.640564][2633:2633] CHIP:TS: Pending Last Known Good Time: 2022-07-21T23:35:45
  [1658929461.640771][2633:2633] CHIP:TS: Previous Last Known Good Time: 2022-07-21T23:35:45
  [1658929461.640822][2633:2633] CHIP:TS: Reverted Last Known Good Time to previous value
 * steps to reproduce
 AP1:
 ./chip-tool generalcommissioning arm-fail-safe 900 1 1 0
 ./chip-tool networkcommissioning remove-network <SSID of AP1> 1 0
 ./chip-tool networkcommissioning add-or-update-wi-fi-network <SSID of AP2> <PSK of AP2> 1 0
 ./chip-tool networkcommissioning connect-network <SSID of AP2> 1 0

AP2:
 ./chip-tool networkcommissioning read networks 1 0
 ./chip-tool generalcommissioning arm-fail-safe 0 0 1 0

AP1:
 ./chip-tool networkcommissioning read networks 1 0
 * system configuration
  EFR32 + RS9116
  EFR32 + WF200

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-07-27:
The expected behavior does not seem right.  As you said, if the DUT is on a different network now, if TH is not on that network it can't get the response.

chirag-silabs, 2022-07-27:
Yup, Should an issue be raised in test plan for this, TC-CNET-4.11

woody-apple, 2022-07-27:
Cert Blocker Review: @cjandhyala to follow up.

woody-apple, 2022-07-27:
Cert Blocker Review: Need to better understand the scope of this, could be a provisional candidate for 7/29

franck-apple, 2022-07-28:
Per discussion with @cjandhyala, this seems to be a test environment issue (test plan issue here: https://github.com/CHIP-Specifications/chip-test-plans/issues/1947). Closing. Please re-open if SDK work is needed.

closed 2022-07-28 by franck-apple