project-chip/connectedhomeip #21314: CHIPDeviceController needs proper support for setting network credentials during a commissioning session

chrisdecenzo, 2022-07-28:
#### Problem
 * In many cases, the selection of network credentials needs to be done asynchronously during the PASE session. For example, after the commissioner determines the network technology of the target (WiFi or Thread) and after a network scan has been done. Often this selection will need user or cloud input, so it needs to be asynchronous.
 * A workaround was quickly added in PR https://github.com/project-chip/connectedhomeip/pull/20766 which includes a pause/resume feature of the AutoCommissioner but a cleaner fix would be to introduce steps in CHIPDeviceController for requesting network credentials. This would allow commissioning apps to avoid needing to create their own AutoCommissioner and pass it in to CHIPDeviceController.

 #### Proposed Solution
* Add optional steps to CHIPDeviceController for obtaining network credentials from its caller.

mrjerryjohns, 2022-08-01:
Can we close this issue @chrisdecenzo ?

bzbarsky-apple, 2022-08-05:
And to be clear, the issue is not "creating your own AutoCommissioner"; the issue is the extra pause API and whatnot, which introduces yet another way of doing async operations via the delegate when we have a perfectly good way already: async call into the delegate.

PR #21876 (2022-08-16): Issue 21314 - add proper step to commissioner for obtaining network credentials
#### Problem
* Fixes #21314 CHIPDeviceController needs proper support for setting network credentials during a commissioning session

#### Change overview
* introduce CommissioningStage.kNeedsNetworkCreds step
* document expectations for scanNetworks callbacks in DevicePairingDelegate

#### Testing
* Tested using Android commissioner
closed 2022-08-16 by woody-apple