project-chip/connectedhomeip #21315: [TC-DGGEN-2.3] fix all cluster app on raspi linux to return correct TestEventTriggersEnabled attribute

cjandhyala, 2022-07-28:
fix all cluster app on raspi linux to return correct TestEventTriggersEnabled attribute. Right now it turns False, hence we can not test TC-DGGEN-2.3 on the raspi platform. TestEventTriggersEnabled is a mandatory attribute in General Diagnostics cluster

cjandhyala, 2022-07-29:
per instructions from @tcarmelveilleux and @Damian-Nordic , I am able to build the all-cluster-app (on raspi linux) without using 'chip_build_libshell=true' flag in the all-cluster app using build command i.e. 'scripts/examples/gn_build_example.sh examples/all-clusters-app/linux/ out/all-clusters-app chip_inet_config_enable_ipv4=false chip_enable_ota_requestor=true' . 

However, I get CONSTRAINT_ERROR when trying to send test-event-trigger command

```
>>> generaldiagnostics test-event-trigger 00112233445566778899aabbccddeeff 0xFFFFFFFFFFFFFFFF 80 0   
[1659042499023] [59872:2730652] CHIP: [TOO] Sending command to node 0x50
[1659042499023] [59872:2730652] CHIP: [CSM] FindOrEstablishSession: PeerId = 85E5BB11F0FDF4A2:0000000000000050
[1659042499023] [59872:2730652] CHIP: [TOO] Sending cluster (0x00000033) command (0x00000000) on endpoint 0
[1659042499023] [59872:2730652] CHIP: [DMG] ICR moving to [AddingComm]
[1659042499023] [59872:2730652] CHIP: [DMG] ICR moving to [AddedComma]
[1659042499024] [59872:2730652] CHIP: [IN] Prepared secure message 0x7f8226fe7658 to 0x0000000000000050 (1) of type 0x8 and protocolId (0, 1) on exchange 14081i with MessageCounter:38960271.
[1659042499024] [59872:2730652] CHIP: [IN] Sending encrypted msg 0x7f8226fe7658 with MessageCounter:38960271 to 0x0000000000000050 (1) at monotonic time: 000000000BD01FBA msec
[1659042499024] [59872:2730652] CHIP: [DMG] ICR moving to [CommandSen]
[1659042499028] [59872:2734042] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:42052470 on exchange 14081i
[1659042499028] [59872:2734042] CHIP: [EM] Found matching exchange: 14081i, Delegate: 0x7f8226fe79b0
[1659042499028] [59872:2734042] CHIP: [EM] Rxd Ack; Removing MessageCounter:38960271 from Retrans Table on exchange 14081i
[1659042499028] [59872:2734042] CHIP: [EM] Removed CHIP MessageCounter:38960271 from RetransTable on exchange 14081i
[1659042499028] [59872:2734042] CHIP: [DMG] ICR moving to [ResponseRe]
[1659042499028] [59872:2734042] CHIP: [DMG] InvokeResponseMessage =
[1659042499028] [59872:2734042] CHIP: [DMG] {
[1659042499028] [59872:2734042] CHIP: [DMG] 	suppressResponse = false, 
[1659042499028] [59872:2734042] CHIP: [DMG] 	InvokeResponseIBs =
[1659042499028] [59872:2734042] CHIP: [DMG] 	[
[1659042499028] [59872:2734042] CHIP: [DMG] 		InvokeResponseIB =
[1659042499028] [59872:2734042] CHIP: [DMG] 		{
[1659042499028] [59872:2734042] CHIP: [DMG] 			CommandStatusIB =
[1659042499028] [59872:2734042] CHIP: [DMG] 			{
[1659042499028] [59872:2734042] CHIP: [DMG] 				CommandPathIB =
[1659042499028] [59872:2734042] CHIP: [DMG] 				{
[1659042499028] [59872:2734042] CHIP: [DMG] 					EndpointId = 0x0,
[1659042499028] [59872:2734042] CHIP: [DMG] 					ClusterId = 0x33,
[1659042499028] [59872:2734042] CHIP: [DMG] 					CommandId = 0x0,
[1659042499028] [59872:2734042] CHIP: [DMG] 				},
[1659042499028] [59872:2734042] CHIP: [DMG] 				
[1659042499028] [59872:2734042] CHIP: [DMG] 				StatusIB =
[1659042499028] [59872:2734042] CHIP: [DMG] 				{
[1659042499028] [59872:2734042] CHIP: [DMG] 					status = 0x87 (CONSTRAINT_ERROR),
[1659042499028] [59872:2734042] CHIP: [DMG] 				},
[1659042499028] [59872:2734042] CHIP: [DMG] 				
[1659042499028] [59872:2734042] CHIP: [DMG] 			},
[1659042499028] [59872:2734042] CHIP: [DMG] 			
[1659042499028] [59872:2734042] CHIP: [DMG] 		},
[1659042499028] [59872:2734042] CHIP: [DMG] 		
[1659042499028] [59872:2734042] CHIP: [DMG] 	],
[1659042499028] [59872:2734042] CHIP: [DMG] 	
[1659042499028] [59872:2734042] CHIP: [DMG] 	InteractionModelRevision = 1
[1659042499028] [59872:2734042] CHIP: [DMG] },
[1659042499028] [59872:2734042] CHIP: [DMG] Received Command Response Status for Endpoint=0 Cluster=0x0000_0033 Command=0x0000_0000 Status=0x87
[1659042499028] [59872:2734042] CHIP: [TOO] Error: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)
[1659042499028] [59872:2734042] CHIP: [DMG] ICR moving to [AwaitingDe]
[1659042499028] [59872:2734042] CHIP: [EM] Sending Standalone Ack for MessageCounter:42052470 on exchange 14081i
[1659042499028] [59872:2734042] CHIP: [IN] Prepared secure message 0x700001ab3588 to 0x0000000000000050 (1) of type 0x10 and protocolId (0, 0) on exchange 14081i with MessageCounter:38960272.
[1659042499028] [59872:2734042] CHIP: [IN] Sending encrypted msg 0x700001ab3588 with MessageCounter:38960272 to 0x0000000000000050 (1) at monotonic time: 000000000BD01FBE msec
[1659042499028] [59872:2722530] CHIP: [TOO] Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)
[1659042499028] [59872:2734042] CHIP: [EM] Flushed pending ack for MessageCounter:42052470 on exchange 14081i
```
Spec Reference 
[image]

tcarmelveilleux, 2022-07-29:
In `8.8.2.3. Incoming Invoke Request Action`:

> - If a data field violates expected constraints, a CommandStatusIB SHOULD be generated with an error status of CONSTRAINT_ERROR.

CONSTRAINT_ERROR is correct here.

SDK and spec are correct. TC-DGEN-2.3 must be amended.

cjandhyala, 2022-07-29:
closing the issue per @tcarmelveilleux comment

closed 2022-07-29 by cjandhyala