project-chip/connectedhomeip #21341: [SQA] TC-CNET-4.9- Network changes are not reverted After the Failsafe in WIFI.

Saravana-kr22, 2022-07-28:
#### Problem
Disarming the failsafe should revert all changes made inside the failsafe. 
But in Ble-wifi Pairing, after setting the failsafe to 0 also the changes are not reverted in TC-CNET-4.9.

**Expected behavior  :**  

> [1658992400.314721][3054:3059] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 1365689526
> [1658992400.314804][3054:3059] CHIP:TOO:   Networks: 1 entries
> [1658992400.314854][3054:3059] CHIP:TOO:     [1]: {
> [1658992400.314884][3054:3059] CHIP:TOO:       NetworkID: 63686970736574757034
> [1658992400.314909][3054:3059] CHIP:TOO:       Connected: TRUE
> [1658992400.314931][3054:3059] CHIP:TOO:      }
> [1658992400.315030][3054:3059] CHIP:EM: Sending Standalone Ack for MessageCounter:184714660 on exchange 54594i

**Actual behavior  :**  

> [1658993130.776793][3171:3176] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 1365689526
> [1658993130.776898][3171:3176] CHIP:TOO:   Networks: 0 entries
> [1658993130.777029][3171:3176] CHIP:EM: Sending Standalone Ack for MessageCounter:156225182 on exchange 13044i

**Note:** In Ble-thread, after setting the failsafe to 0 changes are reverted.

**System configuration:**

- DUT : All-cluster-app ( RPI)
- CNTRL : Chiptool ( RPI)
- Network : Ble-Wifi

**Steps to reproduce :**  

- Step 11 is a disarm of the failsafe, so all network configurations from step 1 to step 10 should be negated
- Step 1 is arm the failsafe
- steps 2 - 10 is to remove WIFI_1ST_ACCESS_POINT, add WIFI_2ND_ACCESS_POINT
- step 11: Disarm the failsafe. This should revert all changes made from step 2-10. So DUT should be back in the pretest state, connected to WIFI_1ST_ACCESS point
- Step 10 - TH connets back to WIFI_1ST_ACCESS_POINT

**Log:**
[cnet_error.txt](https://github.com/project-chip/connectedhomeip/files/9208008/cnet_error.txt)

cjandhyala, 2022-07-29:
@Saravana-kr22 Could you pls attach a complete log

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

Saravana-kr22, 2022-08-01:
@cjandhyala Here is the full log of the test case: [cnet-4.9.txt](https://github.com/project-chip/connectedhomeip/files/9234381/cnet-4.9.txt)

chirag-silabs, 2022-08-02:
@Saravana-kr22 just a query this happens the same in 4.11 as well right?

Saravana-kr22, 2022-08-02:
@chirag-silabs Yes, The same issue happens in TC-CNET-4.11.

chirag-silabs, 2022-08-12:
@Saravana-kr22 are you able to get the connect success message once the DUT is connected to the new network? I am not able to get the success message but my DUT is connected to another network on chip-tool side but I am able to get the connect success response. Any idea what can be the reason

tcarmelveilleux, 2022-09-01:
The fact that ArmFailSafe does not revert is DIFFERENT from the fact that a response is not given. Let's not mix issues

closed 2023-07-18 by Survensa