project-chip/connectedhomeip #21348: Android chip-tool has been broken for over a week

jonsmirl, 2022-07-28:
Clean checkout and build this morning on 8fe47272d2288e81d5480be4254f03079bb74744

Built all-clusters and flashed into ESP32
Built Android app and loaded into Pixel 3a.

Pairing fails...

I (72512) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 0 
E (72522) chip[DL]: Long dispatch time: 416 ms, for event type 7
I (72602) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (72602) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72692) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72692) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183631302 on exchange 37016r
I (72712) chip[ZCL]: OpCreds: Received an AddNOC command
E (72712) chip[ZCL]: OpCreds: Failed AddNOC request with IM error 0x85
I (72722) chip[IN]: Prepared secure message 0x3fcb78e8 to 0xFFFFFFFB00000000 (0)  of type 0x9 and protocolId (0, 1) on exchange 37016r with MessageCounter:207460002.
I (72732) chip[IN]: Sending encrypted msg 0x3fcb78e8 with MessageCounter:207460002 to 0xFFFFFFFB00000000 (0) at monotonic time: 0000000000011C1D msec
I (72742) NimBLE: GATT procedure initiated: indicate; 
I (72752) NimBLE: att_handle=14

[android.log](https://github.com/project-chip/connectedhomeip/files/9209548/android.log)

mrjerryjohns, 2022-08-01:
I don't believe the Android chip-tool is being actively maintained at this point.

jonsmirl, 2022-08-01:
So no more example for anyone writing a based Java controller app to start from?
The tv-app does not work the same way.

This is going to make it much harder for vendors to write their own setup apps.

tatsutaigu, 2022-08-02:
I think they move to their beta program.
[https://developers.home.google.com/matter/get-started](https://developers.home.google.com/matter/get-started)
[https://developer.apple.com/apple-home/](https://developer.apple.com/apple-home/)

jonsmirl, 2022-08-02:
It is more than Android chip-tool being broken. Android chip-tool is broken because the underlying JNI library is also currently broken.  I applied to the Google beta, maybe they can supply me with a working Java library.  Android chip-tool was just an easy way to test if the JNI library was functional.

tatsutaigu, 2022-08-02:
They have already distributed it.
It may include JNI library.
[https://developers.home.google.com/mobile-sdk](https://developers.home.google.com/mobile-sdk)

jonsmirl, 2022-08-02:
I have access to their developer portal now and I think this mobile-sdk is what I need, but it has not been released yet. They are putting the chip library into Google Play Services.  Looks like I have to commission onto their fabric first, and only then will it make another fabric and share over to our our app. Hopefully it will let me send vendor commands to set up our features which are not part of the CHIP spec. I wonder what Apple is going to do since I need to do the same process on IOS.

I'm also not clear on how this will work with us having our app installed on multiple phones in the household. Those other phones need access to our fabric. 

```
Watch a brief overview of how the Google Home Mobile SDK lets you build Android Matter apps with great setup, device sharing, and control experiences.

More details coming soon!
```

hiepverik, 2022-08-09:
I found the root cause of the issue `Failed AddNOC request with IM error 0x85`
Just change https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/src/chip/devicecontroller/ControllerParams.java#L103
From
```
private int controllerVendorId = 0xFFFF;
```
To 
```
private int controllerVendorId = 0xFFF4;
```
or any id below `0xFFF4`

Root cause: https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPVendorIdentifiers.hpp#L60

tatsutaigu, 2022-08-10:
@hiepverik 
Thank you for your information!
I change controllerVendorId = 0xFFFF to 0xFFF4, then, Android app can control ESP32.

andy31415, 2022-09-15:
I believe this can be closed: need to use a valid vendor ID as per spec (test vendors IDs are 0xFFF1 to 0xFFF4 ... real apps should use a real vendor id)

closed 2022-09-15 by andy31415