project-chip/connectedhomeip #21362: [TC-ACT-2.2]TH fails to subscribe StateChanged and ActionFailed event.

Rajashreekalmane, 2022-07-28:
#### Problem
As per Test-Plan, If TH sends a subscribe request to DUT,  DUT should responds with  an event list Report Data Message with the data for specific event,  But DUT responds as  No entry for an event in EventReports list

 * **expected behavior**:-Report Data Message with the data for specific event  
 * **actual behavior:**- No entry for that event in EventReports list
 * **steps to reproduce:-**

1.  ./chip-tool interactive start
2. bridgedactions subscribe-event   action-failed  2 30 1 0
3.  bridgedactions subscribe-event  state-changed      2 30 1 0

 **system configuration:-**
 system configuration
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID:29873290196796fbf23e28083163f8565bb479ba
Dut application: bridge-app

**Test-plan Issue:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/actions.adoc#tc-act-2-2-primary-functionality-with-server-as-dut
**PFA logs below:-**
[TC-ACT-2.2.txt](https://github.com/project-chip/connectedhomeip/files/9212948/TC-ACT-2.2.txt)

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

leorozendaal, 2022-08-05:
@Rajashreekalmane  @woody-apple   My assumption is that reaction of this DUT is that no events of this type have been issued so far and therefore it's effectively "nothing has happened so far" confirmation.
@bzbarsky-apple @mrjerryjohns   is this expected behavior (from spec/SDK point of view) when setting up an event subscription?

bzbarsky-apple, 2022-08-05:
Yes, it is.  If there are no events of that type queued, you get an empty list back.

leorozendaal, 2022-08-05:
thanks @bzbarsky-apple 
@Rajashreekalmane  so it seems what you get is what you would expect

Rajashreekalmane, 2022-08-09:
@leorozendaal Thanks, Verification steps updated accordingly.

closed 2022-08-09 by Rajashreekalmane