project-chip/connectedhomeip #21391: [ESP32] Where are Discriminator and Setup-pin-code in menuconfig?

tatsutaigu, 2022-07-29:
#### Problem
I cannot find discriminator and setup pin code setting in menuconfig.
Where is it?
How do I change it?
[https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32#commissioning-over-ble](https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32#commissioning-over-ble)

I tried `grep "f00" sdkconfig`, `grep "3840" sdkconfig` and `grep "20202021" sdkconfig`, but not found.

connectedhomeip : commit ae2ca6d22e6594614df712fec3becad8e6ef2c6c (HEAD -> master, origin/master, origin/HEAD)
ESP-IDF : v4.4.1
device : ESP32-C3-DevkitM-1

bzbarsky-apple, 2022-07-29:
@shubhamdp @dhrishi

dhrishi, 2022-07-29:
@tatsutaigu In the end products, these parameters won't be configurable through menuconfig but would be fixed an pre-configured in the device. As we are getting closer to production, the support for this was removed on purpose. You can check the sub-section "Using ESP32 Factory Data Provider (Optional)" [here](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/esp32#building-the-example-application) to configure the values of your choice. If the values are not set using the factory data provider, the default ones are used (discriminator 3840 and setup pincode 20202021) from the common SDK code.

And, the documentation you pointed out seems to be stale and we will update it right away. Thanks for reporting!

shubhamdp, 2022-07-29:
I have removed the those stale pointers in #21340

tatsutaigu, 2022-07-29:
@dhrishi and @shubhamdp 

Thank you for your prompt reply and action.
I understand it.

closed 2022-07-29 by tatsutaigu