project-chip/connectedhomeip #21395: [EFR32]  TC-CNET-4.4 failed on scanning the particular AP details

shgutte, 2022-07-29:
#### Problem
In TC-CNET-4.4 need to scan the particular SSID but sometimes it returns the empty structure and sometimes we are getting the values for the SSID.

Actual output:
[1659002761.855917][26059:26064] CHIP:TOO:   ScanNetworksResponse: {
[1659002761.856046][26059:26064] CHIP:TOO:     networkingStatus: 0
[1659002761.856121][26059:26064] CHIP:TOO:     wiFiScanResults: 0 entries
[1659002761.856404][26059:26064] CHIP:TOO:    }

Expected output:
[1659002756.133405][26053:26058] CHIP:TOO:   ScanNetworksResponse: {
[1659002756.134124][26053:26058] CHIP:TOO:     networkingStatus: 0
[1659002756.134189][26053:26058] CHIP:TOO:     wiFiScanResults: 1 entries
[1659002756.134256][26053:26058] CHIP:TOO:       [1]: {
[1659002756.134298][26053:26058] CHIP:TOO:         Security: 2
[1659002756.134341][26053:26058] CHIP:TOO:         Ssid: 6D61747465725F74656E6461
[1659002756.134380][26053:26058] CHIP:TOO:         Bssid: 502B73B61B51
[1659002756.134416][26053:26058] CHIP:TOO:         Channel: 10
[1659002756.134453][26053:26058] CHIP:TOO:         WiFiBand: 0
[1659002756.134489][26053:26058] CHIP:TOO:         Rssi: -74
[1659002756.134524][26053:26058] CHIP:TOO:        }
[1659002756.134563][26053:26058] CHIP:TOO:    }

Tested on sve branch commit ID: c803dab4ec3ad97035e02332a86e057b58f5b7c5

App used - lock app
Platform - Chip-tool - RPI-4, 8GB RAM
DUT - EFR32 + WF200
Network - Ble-wifi
Endpoint: 1

Logs: 
[TC-CNET-4.4-2-chip-toollog.txt](https://github.com/project-chip/connectedhomeip/files/9216556/TC-CNET-4.4-2-chip-toollog.txt)
[TC-CNET-4.4-2-device-log.txt](https://github.com/project-chip/connectedhomeip/files/9216557/TC-CNET-4.4-2-device-log.txt)

cjandhyala, 2022-07-29:
@shgutte if you provisioned the device over IP transport, the scan results shown in ScanNetworksResponse will be 0. pls chk which transport you are provisioning the device with. If you provision over Wifi it will produce results.

bzbarsky-apple, 2022-07-29:
@cjandhyala This is not a Linux thing; it's an EFR32 device.  And the point, if you look in the logs, is that an undirected scan is succeeding but a directed scan is not returning useful data.

@shgutte Looking at the logs, the directed scan is scanning for the SSID "matter_tenda", right?  I don't see that ssid in the listing the undirected scan in the log produced, so the most probable thing going on is that the DUT is not in fact seeing that SSID for some reason?

closed 2023-02-04 by stale[bot]