project-chip/connectedhomeip #21400: Need tool support for generating elements

sowmyassp, 2022-07-29:
**Issue :** 
**Need tools supported for  the following** 

1. For generating invalid NOCSR Information
2. To extracts the TLV-encoded nocsr_elements_message from nocsr_elements 
3. For generating the NOC
4. For generating new NOC is generated from the NOCSR returned in csr_update and is signed by new ICA
5. For generating the new ICAC is generated and signed by new_root_key

**Test-cases is affected due to the above needs**

1. TC-OPCREDS-3.1
2. TC-OPCREDS-3.4
3. TC-OPCREDS-3.5
4. TC-DA-1.5
5. TC-DA-1.6

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

Rajashreekalmane, 2022-12-12:
TC-DA-1.2]-
Support required for a specific decode tool in chip-cert to decode the signature & construct a new message and required a Support for certification_declaration CMS enveloped is required.

Test-plan Reference: 
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/deviceattestation.adoc#tc-da-1-2-device-attestation-request-validation-dut-commissionee