project-chip/connectedhomeip #21405: [SQA] TC-CNET-4.3 LastNetworkingStatus & LastNetworkID Returns Null value.

Survensa, 2022-07-29:
#### Problem
In TC-CNET-4.3 In Ethernet Pairing the Network Attribute list has one entry, But while executing the LastNetworkingStatus & LastNetworkID command. It throws the Null value as the Output. It is not the Expected Outcome.

**Note:**  In Ble-Wifi & Ble-Thread pairing , It gives the Excepted Outcome.

**Excepted Outcome:**

> [1654076802.288579][33669:33674] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0006 DataVersion: 609053543
[1654076802.288708][33669:33674] CHIP:TOO:   LastNetworkID: 6368697035

**Actual Outcome:** 

> [1659077129.806911][2906:2911] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0006 DataVersion: 832817014
[1659077129.806983][2906:2911] CHIP:TOO:   LastNetworkID: null
[1659077129.807154][2906:2911] CHIP:EM: Sending Standalone Ack for MessageCounter:523769 on exchange 18282i 

**Steps to reproduce:** 

1. `./chip-tool networkcommissioning read networks 1 0`

> [1654076774.294361][33659:33664] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 609053543
[1654076774.294505][33659:33664] CHIP:TOO:   Networks: 1 entries
[1654076774.294625][33659:33664] CHIP:TOO:     [1]: {
[1654076774.294679][33659:33664] CHIP:TOO:       NetworkID: 6368697035
[1654076774.294726][33659:33664] CHIP:TOO:       Connected: TRUE
[1654076774.294774][33659:33664] CHIP:TOO:      }

2. `./chip-tool networkcommissioning read last-networking-status 1 0`
3. `./chip-tool networkcommissioning read last-network-id 1 0`

**System configuration:** 

- Cntl: RPI-Chiptool
- DUT: RPI-All-cluster-app
- Network:Ethernet

**Log:**

[cnet-4.3.txt](https://github.com/project-chip/connectedhomeip/files/9219673/cnet-4.3.txt)

bzbarsky-apple, 2022-07-29:
What the spec for LastNetworkingStatus says is:

> This attribute SHALL indicate the status of the last attempt either scan or connect to an operational network, using this interface, whether by invocation of the ConnectNetwork command or by autonomous connection after loss of connectivity or during initial establishment. If no such attempt was made, or no network configurations exist in the Networks attribute, then this attribute SHALL be set to null.

In this case there was no ConnectNetwork command.  I'm not sure whether there was "autonomous connection" going on here or not....  So it's possible that null is in fact a correct value to return.

LastNetworkID has similar language.

We need to figure out whether this is a bug in the Ethernet driver or whether the test expectations need adjusting.

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today. Possible this is an error in the test case as well.

Survensa, 2022-08-01:
> What the spec for LastNetworkingStatus says is:
>
>
> In this case there was no ConnectNetwork command. I'm not sure whether there was "autonomous connection" going on here or not.... So it's possible [...]

@bzbarsky-apple There is bug only with ethernet driver because it works fine with `Blw-wifi` and `Ble-thread`

bzbarsky-apple, 2022-08-02:
@Survensa Wi-Fi and Thread do ConnectNetwork during commissioning.  Ethernet does not.

Survensa, 2022-08-03:
Raised test plan issue: https://github.com/CHIP-Specifications/chip-test-plans/issues/1993

closed 2022-08-03 by Survensa