project-chip/connectedhomeip #21406: [TC-LVL-4.1] TH fails to Move the DUT up.

Rajashreekalmane, 2022-07-29:
#### Problem
As per test-plan in test step 1b TH sends a move command to DUT with the move mode field set to 0x00 (move up) and the rate field set to 0x20 (32 units per second) DUT should responds as Success, while verifying move command by sending currentlevel attribute responds as 1 instead of 0xFE.

 * expected behavior:-  DUT should respond 0xFE  
 * actual behavior:-DUT is responding as 1 
 * steps to reproduce:-
 ./chip-tool onoff on 1 1
 ./chip-tool levelcontrol write  options 0x00 1 1
./chip-tool levelcontrol read min-level 1 1
 ./chip-tool levelcontrol move-to-level-with-on-off  1 0 0 0 1 1
./chip-tool levelcontrol read max-level 1 1
./chip-tool levelcontrol move 0x00 0x20 0 0 1 1
./chip-tool levelcontrol read current-level 1 1

 * system configuration
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID: cf37a25b6e96436d3d34a8967c2eb7b3053e0a50
Dut application: all-clusters-app

Test-plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/levelcontrol.adoc#tc-lvl-4-1-move-verification-dut-as-server

PFA log below:-
[[TC-LVL-4.1].txt](https://github.com/project-chip/connectedhomeip/files/9219737/TC-LVL-4.1.txt)
[TC-LVL-4.1--DUT.txt](https://github.com/project-chip/connectedhomeip/files/9219740/TC-LVL-4.1--DUT.txt)

leorozendaal, 2022-07-29:
(discussing with @Rajashreekalmane  via slack)

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

leorozendaal, 2022-07-31:
root cause (likely) is the use of parameter level=1 in MoveToLevelWithOnOff which will not turn on the device (by spec).
Expected to be resolved by rework of testplan => https://github.com/CHIP-Specifications/chip-test-plans/pull/1965

closed 2022-08-01 by cjandhyala