project-chip/connectedhomeip #21407: [TC-LVL-5.1] TH fail to step up  the DUT at step 2a.

Rajashreekalmane, 2022-07-29:
#### Problem
As per test-plan test step 2a "TH sends a Step command to DUT SERVER, with the step mode field set to 0x00 (step up), the step size field set to 0x40 (64 units) and the transition time field set to 0x14 (2s)" DUT should responds with success, while verifying step command by reading  CurrentLevel attribute, DUT responds currentlevel  as 1 instead 0x40 (64 units)
 
* expected behavior:-DUT should respond currentlevel as 64 units(0x40) 
 * actual behavior:-DUT responds currentlevel  as 1 
 * steps to reproduce

 **system configuration**
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : IP and wifi
Commit-ID: cf37a25b6e96436d3d34a8967c2eb7b3053e0a50
Dut application: all-clusters-app

Steps to reproduce:-
./chip-tool onoff on 1 1
 ./chip-tool levelcontrol write  options 0x00 1 1
./chip-tool levelcontrol read min-level 1 1
 ./chip-tool levelcontrol  move-to-level-with-on-off 1 0 0 0 1 1
  ./chip-tool levelcontrol read current-level 1 1
./chip-tool levelcontrol  step  0 64 20 1 1 1 1
 ./chip-tool levelcontrol read current-level 1 1

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/levelcontrol.adoc#tc-lvl-5-1-step-verification-dut-as-server

**PFA log below:-**
[TC-LVL-5.1.txt](https://github.com/project-chip/connectedhomeip/files/9219813/TC-LVL-5.1.txt)
[TC-LVL-5.1 DUT.txt](https://github.com/project-chip/connectedhomeip/files/9219816/TC-LVL-5.1.DUT.txt)

leorozendaal, 2022-07-29:
(discussing with @Rajashreekalmane  via slack)

woody-apple, 2022-07-29:
Cert Blocker Review: Marking this as 'needs provisional' unless we have a resolution to this issue today.

leorozendaal, 2022-07-31:
root cause (likely) is the use of parameter level=1 in MoveToLevelWithOnOff which will not turn on the device (by spec).
Expected to be resolved by rework of testplan => https://github.com/CHIP-Specifications/chip-test-plans/pull/1965

leorozendaal, 2022-07-31:
root cause (likely) is the use of parameter level=1 in MoveToLevelWithOnOff which will turn off the device (by spec).
Expected to be resolved by rework of testplan => https://github.com/CHIP-Specifications/chip-test-plans/pull/1965

closed 2022-08-01 by cjandhyala