project-chip/connectedhomeip #21410: Bridge-functionality - The keypress 2 and 4 is aborting the DUT

sowmyassp, 2022-07-29:
**Issue :**
The keypress 2 and 4 are aborting the DUT
[image]

**Commit used** : https://github.com/project-chip/connectedhomeip/commit/29873290196796fbf23e28083163f8565bb479ba
**Chip-tool** - RPI-4, 8GB RAM
**DUT** - RPI - RPI-4, 8GB RAM
**Platform** : linux platform
**Dut application**: bridge-app

bzbarsky-apple, 2022-07-29:
On Linux, https://github.com/project-chip/connectedhomeip/pull/21287 will fix this, by holding the right locks.

On Mac, we will need a bit more work.  But given this issue is for RPI, https://github.com/project-chip/connectedhomeip/pull/21287 should be enough.

sowmyassp, 2022-07-29:
Ok then I will follow up with 21287 PR on linux, and close this isue here

closed 2022-07-29 by sowmyassp