project-chip/connectedhomeip #21453: SubscriptionCallback::ReportData needs to capture its callbacks in block vars.

bzbarsky-apple, 2022-07-29:
#### Problem
SubscriptionCallback::ReportData  (in MTRBaseDevice.mm) does this:
```
    if (mAttributeReportCallback && attributeReports.count) {
        dispatch_async(mQueue, ^{
            mAttributeReportCallback(attributeReports);
```

but by the time the async block runs, `mAttributeReportCallback` can have become null, right?  And in general, this is doing racy access to the callback....

We have crashes on that call to mAttributeReportCallback that I bet are due to this.

 #### Proposed Solution
The callback should be getting captured into a block var, like the attributeReports are.

closed 2022-08-01 by woody-apple

aajain-com, 2022-08-02:
@bzbarsky-apple @woody-apple  this seems very similar to https://github.com/project-chip/connectedhomeip/issues/21142 that I filed given that  `SubscriptionCallback::OnReportEnd` ends up calling `ReportData()` as per the code. WDYT?

bzbarsky-apple, 2022-08-02:
@aajain-com Yes, I expect #21142 is the same issue, with `mEventReportCallback` having become null.