project-chip/connectedhomeip #21468: iOS 16 beta 4 commissioning aborts with "Certificate not yet valid"

plan44, 2022-07-30:
#### Context

Commissioning example apps from connectedhomeip (in my case: bridge-app) with iOS was possible following the guide document [docs/guides/darwin.md](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md) using TE9 checkout.

#### Problem

Since iOS 16 beta 4 was released on 2022-07-27, commissioning always fails, logging (in the commissioner, in the iOS device console) the reason as **Certificate not yet valid** (using the TE9 checkout or any of the later versions that worked fine with beta 3 for the commissionee):

```
default	11:37:36.649367+0200	homed	🔵 [1659087456490] [956:356383] CHIP: [CTL] Device connection failed. Error ../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:444: CHIP Error 0x0000004F: Certificate not yet valid
default	11:37:36.649451+0200	homed	🔵 [1659087456491] [956:356383] CHIP: [CTL] Error on commissioning step 'FindOperational': '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:444: CHIP Error 0x0000004F: Certificate not yet valid'
error	  11:37:36.649477+0200	homed	🔴 [1659087456491] [956:356383] CHIP: [CTL] Failed to perform commissioning step 18
default	11:37:36.649534+0200	homed	🔵 [1659087456491] [956:356383] CHIP: [CTL] Going from commissioning step 'FindOperational' with lastErr = '../../../../../../../../Sources/CHIPFramework/connectedhomeip/src/credentials/CHIPCert.cpp:444: CHIP Error 0x0000004F: Certificate not yet valid' -> '
```

In the commissionee, the log shows a `received error (protocol coce 2) during pairing process` with a `CHIP_ERROR_INVALID_CASE_PARAMETER` (0x54) error.

Now the question is: is this "not yet" just a few days in the future due perhaps due to slight misalingnment of expected and actual release of these apparently new certificates in the iOS betas? Which could mean the fix is just waiting a few days?

Unfortunately it is not clear to me which would be the relevant certificate that seems to a have a future validity date in the version deployed with iOS beta 4.

Any idea when and how one can continue testing with iOS?

bzbarsky-apple, 2022-08-01:
Hmm.  This sounds a lot like https://github.com/project-chip/connectedhomeip/issues/20302

Is it possible that the timezone on the iOS device changed around the same time as it was updated to the beta?  What timezone is the iOS device using in this case?

bzbarsky-apple, 2022-08-01:
Also, are you using the "TE9" version of iOS CHIPTool that you compile yourself?  That is, are you using an iOS CHIPTool that does not have the fix for https://github.com/project-chip/connectedhomeip/issues/20302 in it?

plan44, 2022-08-02:
Thx for the useful info, @bzbarsky-apple, Indeed, that sounds much like #20302

However, I am not using chip-tool, but trying to test my bridge-app derivative against the iOS built-in stack as described in the [darwin.md guide document](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md).

So what I was trying to find out is what might have changed between the chip version in beta3 (TE9 if the guide is right) and the one in beta4.

Based on your info, my guess would be that iOS 16 beta4 currently deploys something between June 7th (8cc291d) when *NotBefore* checks were implemented first at all, but *before* July 13th (e7915ea) when you fixed it, so it is affected by #20302.

**In fact, setting my iPad's timezone to UTC (it actually is CEST, which is 2 h ahead of UTC) solves the problem 😄**

That's fine for now, I hope beta 5 will have e7915ea included.

bzbarsky-apple, 2022-08-03:
@plan44 Thank you for following up!  Closing as duplicate of https://github.com/project-chip/connectedhomeip/issues/20302

I don't know what eversion of the Matter SDK will be in beta 5, but I can try to find out if that would be useful.  Let me know, please.

closed 2022-08-03 by bzbarsky-apple

plan44, 2022-08-03:
> I don't know what eversion of the Matter SDK will be in beta 5, but I can try to find out if that would be useful. Let me know, please.

That would certainly be very useful info for anyone testing against the iOS built-in stack! 

Now that there *is* a very useful [guide document](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md) (it's only 2 weeks old, thanks @woody-apple and @carol-apple for adding it), that could be a nice place to record the latest beta's (or the nearest fork point's) SHA.