project-chip/connectedhomeip #21485: Unclear behavior during operational discovery

doru91, 2022-08-01:
#### Problem
There are times when the "on-network" commissioning of the linux ota-provider fails.

When it fails, it look like the Sigma1 doesn't reach the ota-provider (full chip-tool logs in "chip_tool_logs_not_ok_blocked_after_sigma1.txt"): 
```
[1659348782.972155][1772:1779] CHIP:SC: Sent Sigma1 msg
[1659348782.972224][1772:1779] CHIP:CTL: OperationalDeviceProxy[1:0000000000000001]: State change 3 --> 4
[1659348789.101495][1772:1779] CHIP:EM: Retransmitting MessageCounter:72852179 on exchange 42517i Send Cnt 1
[1659348789.101696][1772:1779] CHIP:IN: Sending unauthenticated msg 0xffff78009ef8 with MessageCounter:72852179 to 0  
```

This is confirmed if I look in the tcpdump logs on rpi (capture_nok.pcap): I see only the messages from PASE. 

 #### Proposed Solution
If I set down the br-05e0e23f6237 interface then CASE is successful (see capture_ok.pcap). My guess is that CASE fails when br-05e0e23f6237 is up because chip-tool resolves the ota provider to an address which  is not reachable anymore (fd00:6d61:7474:6572::1) - see also "ip_a_s.txt".

Shouldn't the node retry CASE with a different address according to Section 4.3.2.6 (Performance considerations) of the Specification? I see that the "Additional Records" field contains also some reachable LL IPv6 addresses that could be used (e.g.: the same LL IPV6 address that was used for PASE).

Nodes SHOULD cache the last-known IPv6 address and port for each peer Node with which they
interact from their SRV record resolved using DNS-SD, to save the cost of a run-time network
lookup operation when not needed. When the IPv6 address and port for a peer Node is not
known, or an attempt to communicate with a peer Node at its last-known IPv6 address and port
does not appear to be succeeding within the expected network round-trip time (i.e., the
retransmission timeout value for the first message packet) a Node SHOULD then perform a runtime
discovery in parallel, to determine whether the desired Node has acquired a new IPv6
address and/or port [RFC 8305].

[Operational_discovery_ota.zip](https://github.com/project-chip/connectedhomeip/files/9233185/Operational_discovery_ota.zip)

bzbarsky-apple, 2022-08-01:
> Shouldn't the node retry CASE with a different address according to Section 4.3.2.6

That section does not say "retry with a different address".  It says "check what the address is now and retry with the new address if it has changed".  And we do do that if we had a CASE session established and we send a message and it does not get an ack: we re-resolve the address for the node and update the CASE session to the newly resolved address.

But the situation you're running into is not "we have a stale address cached".  It's "we did a resolve, and there were multiple advertisements, some of them stale because of some _other_ caches in the DNS system, and the one we decided to use is stale", right?

> because chip-tool resolves the ota provider to an address which is not reachable anymore 

Why is it being advertised?  Was it reachable at some point and then the network configuration changed?

> This seems to be a regression as I had no issue in commissioning the ota-provider until updating to the latest master.

@doru91 Are you able to bisect to figure out what changeset introduced the regression?  That would be very helpful in terms of figuring out what might have changed here.

mrjerryjohns, 2022-08-01:
Looking at your logs, `fd00:6d61:7474:6572::1` should indeed be reachable since that looks like the IP address of the BR from the 'thread mesh side'. The BR has a route setup that should forward all traffic to `fd00:6d61:7474:6572::` to the BR interface.

Is ip6 forwarding enabled on the BR? If it isn't, this won't work.

Standard scripts to do this already [exist](https://github.com/openthread/ot-br-posix/blob/main/script/_ipforward) in OTBR.

doru91, 2022-08-16:
In the end I discovered that the issue on my side because I was using the wrong backbone interface for the otbr. This can be closed, thanks.

closed 2022-08-16 by doru91