project-chip/connectedhomeip #21492: Missing `nullptr` checks in `SetUpCodePairer::OnPairingComplete`

dtodor, 2022-08-01:
This code crashes if the pairing delegate has not been set before starting pairing mit the `SetUpCodePairer `.

https://github.com/project-chip/connectedhomeip/blob/cedcce07b318745187f3fdaec165929cd68e6262/src/controller/SetUpCodePairer.cpp#L439

https://github.com/project-chip/connectedhomeip/blob/cedcce07b318745187f3fdaec165929cd68e6262/src/controller/SetUpCodePairer.cpp#L452

mrjerryjohns, 2022-08-01:
Seems like that is the only code-path in `SetUpCodePairer` that doesn't actually check for a valid delegate pointer.

@bzbarsky-apple any specific reason why?

bzbarsky-apple, 2022-08-01:
I am not aware of a specific reason.  `DeviceCommissioner::OnSessionEstablished` does not have a null-check either, and this code is just handling the call from that....

We should likely just add checks in both places, though it's odd to be pairing something without a way to know when pairing is done.

closed 2022-08-02 by woody-apple