project-chip/connectedhomeip #21493: CHIPDeviceController-JNI.cpp incorrect JNIEnv method call for newly added methods

robertfarnum, 2022-08-01:
#### Problem

The following newly added JNI methods use the incorrect JNIEnv method (CallIntMethod):

    bool attemptNetworkScanWiFi        = env->CallIntMethod(controllerParams, getAttemptNetworkScanWiFi);
    bool attemptNetworkScanThread      = env->CallIntMethod(controllerParams, getAttemptNetworkScanThread);

should be (env->CallBooleanMethod):

   bool attemptNetworkScanWiFi        = env->CallBooleanMethod(controllerParams, getAttemptNetworkScanWiFi);
   bool attemptNetworkScanThread      = env->CallBooleanMethod(controllerParams, getAttemptNetworkScanThread);

<what's wrong or missing, please include any applicable:
 * expected behavior
 App does not crash
 * actual behavior
 App crashes on first call above
 * steps to reproduce
 In Android CHIPTool attempt to "PROVISION CHIP DEVICE WITH WI-FI:
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

mrjerryjohns, 2022-08-01:
FYI @chrisdecenzo, since he made that original change.

chrisdecenzo, 2022-08-02:
Working on a fix for this now.

jonsmirl, 2022-08-03:
This problem with boolean is obscuring another problem down deeper...
If the JNI library is working correctly, Android chip-tool should be able to commission devices.
```
I (72512) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 0
E (72522) chip[DL]: Long dispatch time: 416 ms, for event type 7
I (72602) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14
I (72602) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72692) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (72692) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:183631302 on exchange 37016r
I (72712) chip[ZCL]: OpCreds: Received an AddNOC command
--->>>> E (72712) chip[ZCL]: OpCreds: Failed AddNOC request with IM error 0x85
I (72722) chip[IN]: Prepared secure message 0x3fcb78e8 to 0xFFFFFFFB00000000 (0) of type 0x9 and protocolId (0, 1) on exchange 37016r with MessageCounter:207460002.
I (72732) chip[IN]: Sending encrypted msg 0x3fcb78e8 with MessageCounter:207460002 to 0xFFFFFFFB00000000 (0) at monotonic time: 0000000000011C1D msec
I (72742) NimBLE: GATT procedure initiated: indicate;
I (72752) NimBLE: att_handle=14
```

jonsmirl, 2022-08-03:
It should be possible to write some small Java apps to do minimal testing of the JNI library during CI. That would catch breakage like this immediately.

robertfarnum, 2022-08-03:
@chrisdecenzo In the AndroidDeviceControlWrapper.cpp:

CHIP_ERROR err = JniReferences::GetInstance().FindMethod(env, mJavaObjectRef, "onCommissioningStatusUpdate",
"(JLJAVA/LANG/STRING;I)V", &onCommissioningStatusUpdateMethod);

should be:

CHIP_ERROR err = JniReferences::GetInstance().FindMethod(env, mJavaObjectRef, "onCommissioningStatusUpdate",
"(JLjava/lang/String;I)V", &onCommissioningStatusUpdateMethod);

chrisdecenzo, 2022-08-03:
My followup PR should address this: https://github.com/project-chip/connectedhomeip/pull/21522

jonsmirl, 2022-08-03:
Does the CI environment include a JRE allowing small test apps?

jonsmirl, 2022-08-05:
It's get a lot further binding, but still can't finish.
error in onCommissioningStatusUpdate(JLjava/lang/string;I)V

~~~
08-05 00:57:27.492 10080 10101 D IN      : Sending encrypted msg 0x7bfaee37c8 with MessageCounter:218953824 to 0xFFFFFFFB00000000 (0) at monotonic time: 000000000B749171 msec
08-05 00:57:27.492 10080 10101 D DL      : Received SendWriteRequest
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579] JNI DETECTED ERROR IN APPLICATION: JNI NewByteArray called with pending exception java.lang.NoSuchMethodError: no non-static method "Lchip/devicecontroller/ChipDeviceController;.onCommissioningStatusUpdate(JLjava/lang/string;I)V"
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void chip.platform.AndroidChipPlatform.handleIndicationReceived(int, byte[], byte[], byte[]) (AndroidChipPlatform.java:-2)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void chip.platform.AndroidBleManager$1.onCharacteristicChanged(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic) (AndroidBleManager.java:126)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void com.google.chip.chiptool.bluetooth.BluetoothManager$getBluetoothGattCallback$1.onCharacteristicChanged(android.bluetooth.BluetoothGatt, android.bluetooth.BluetoothGattCharacteristic) (BluetoothManager.kt:156)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void android.bluetooth.BluetoothGatt$1$8.run() (BluetoothGatt.java:490)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void android.bluetooth.BluetoothGatt.runOrQueueCallback(java.lang.Runnable) (BluetoothGatt.java:823)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void android.bluetooth.BluetoothGatt.access$200(android.bluetooth.BluetoothGatt, java.lang.Runnable) (BluetoothGatt.java:47)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at void android.bluetooth.BluetoothGatt$1.onNotify(java.lang.String, int, byte[]) (BluetoothGatt.java:484)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at boolean android.bluetooth.IBluetoothGattCallback$Stub.onTransact(int, android.os.Parcel, android.os.Parcel, int) (IBluetoothGattCallback.java:315)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at boolean android.os.Binder.execTransactInternal(int, long, long, int, int) (Binder.java:1184)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]   at boolean android.os.Binder.execTransact(int, long, long, int) (Binder.java:1143)
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579] 
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]     in call to NewByteArray
08-05 00:57:27.493 10080 10101 F e.chip.chiptoo: java_vm_ext.cc:579]     from void chip.platform.AndroidChipPlatform.handleIndicationReceived(int, byte[], byte[], byte[])
08-05 00:57:27.599  3050  3050 I AnimationController: AnimationController.animateTo():91 animateTo: to: 2.0, duration: PT4S

~~~

robertfarnum, 2022-08-05:
That should be: `onCommissioningStatusUpdate(JLjava/lang/String;I)V`  Capital `S` in String

jonsmirl, 2022-08-05:
If I fix onCommissioningStatusUpdate, the AddNOC fix has not landed yet.

I (30699) esp_matter_command: Received command 0x0006 for enpoint 0x0000's cluster 0x0000003E
I (30699) chip[ZCL]: OpCreds: Received an AddNOC command
E (30709) chip[ZCL]: OpCreds: Failed AddNOC request with IM error 0x85

andy31415, 2022-09-15:
I believe this is fixed by #21980

closed 2022-09-15 by andy31415