project-chip/connectedhomeip #21517: When will Android API support firmware update?

TheWalkingDead1024, 2022-08-02:
When will Android API support firmware update?

TheWalkingDead1024, 2022-11-22:
@bzbarsky-apple any update？

bzbarsky-apple, 2022-11-30:
@TheWalkingDead1024 I'm not sure why you're asking me about an _Android_ issue....

srdantas, 2023-05-10:
any updates about this?

TheWalkingDead1024, 2023-08-17:
> @TheWalkingDead1024 I'm not sure why you're asking me about an _Android_ issue....

Thank you for your response. I apologize for any confusion.

Would you be able to point me in the direction of the team member who handles the Android development? I have this issue related to the Android part of the project and would really appreciate any guidance you could provide.

Thank you so much for your time and assistance.

TheWalkingDead1024, 2023-08-18:
@yunhanw-google hi, could you please have look at this issue?

yunhanw-google, 2023-11-08:
Hi @TheWalkingDead1024 we have landed the support for Android OTA/BDX functionality in sdk side, please try the latest code, and use android chip-tool to experiment this functionality, feel free to ping me directly in slack, my slack id is [Google]yunhanw, thanks

closed 2023-11-08 by yunhanw-google