project-chip/connectedhomeip #21520: source bootstrap.sh cloning unnecessary submodules

shubhamdp, 2022-08-02:
#### Problem
- Clone chip repo to depth 1
    ```
    git clone --depth 1 https://github.com/project-chip/connectedhomeip.git /tmp/connectedhomeip
    ```

- Shallow clone the submodules with depth 1 for specific platform
    ```
    scripts/checkout_submodules.py --shallow --platform esp32
    ```

- Below is the output for `source scripts/bootstrap.sh`, and it is deep cloning the submodules
```
$ source scripts/bootstrap.sh
Cloning into '/private/tmp/chip/third_party/bouffalolab/bl602_sdk/repo'...
....
```

 #### Proposed Solution
- It should not clone the submodules or at least do a shallow copy

dhrishi, 2022-08-02:
@mspang Can you please help?

Damian-Nordic, 2022-08-02:
I guess `bootstrap.sh` could simply call `git submodule update` (without `--init`) if `third_party/pigweed/repo/pw_env_setup/util.sh` already exists. That way it would preserve the user's choice on which submodules should be fetched.

shubhamdp, 2022-08-02:
@Damian-Nordic This would be happening in CI as well right?

Damian-Nordic, 2022-08-02:
CI uses `activate.sh` which doesn't do any submodule update if the mentioned `util.sh` file exists.

shubhamdp, 2022-08-02:
Got it, Shall I put up a PR?

Damian-Nordic, 2022-08-02:
If @mspang is OK with that, I don't have any objections :).

Damian-Nordic, 2022-08-02:
Note `--init` is probably needed if `[ ! -f "$_CHIP_ROOT/third_party/pigweed/repo/pw_env_setup/util.sh" ]`. Alternatively, we could simply edit this line to:
```
git submodule update --init third_party/pigweed/repo
```
to only update the submodule that bootstrap depends on.

Damian-Nordic, 2023-02-05:
This is already resolved by https://github.com/project-chip/connectedhomeip/pull/24383

closed 2023-02-05 by Damian-Nordic