project-chip/connectedhomeip #21521: Apple Homekit can't control device when device reset.

shchen-Lab, 2022-08-02:
Hi
I refer apple homekit user guide， this guide can add commission and control my device. My device is BL602 Lighting  example firmware.  This device code commit id  8638802e06fce1501c5c73b40a1d4eae98d076cb base interop_testing_te9 branch.  It can't control by homekit app.
[image]
[image]

[https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md](url)

bzbarsky-apple, 2022-08-02:
When you say "device reset" do you mean power cycle or factory reset?

For the latter, I would not expect it to be controllable anymore.

For the former, I would expect it to work after a single error now that https://github.com/project-chip/connectedhomeip/pull/21504 is in.  At least once that's picked up by the Home app.

closed 2022-08-02 by bzbarsky-apple

shchen-Lab, 2022-08-03:
@bzbarsky-apple   I mean power cycle reset .