project-chip/connectedhomeip #21525: [BRBINFO-2.3]-Response Failure: IM Error 0x0000057F: General error: 0x7f (UNSUPPORTED_ENDPOINT)

revanthb28, 2022-08-02:
#### Problem
Hi All,

We are seeing an issue in BRBINFO-2.3 step 6 case while we are using Hub as the DUT with  TH 1.3

command used

./chip-tool bridgeddevicebasic read node-label 1 3

Issue:
Response Failure: IM Error 0x0000057F: General error: 0x7f (UNSUPPORTED_ENDPOINT)

 * expected behavior
 *[1657696463.081741][15476:15481] CHIP:TOO: Endpoint: 3 Cluster: 0x0000_0039 Attribute 0x0000_0005 DataVersion: 2577979325
[1657696463.081791][15476:15481] CHIP:TOO:   NodeLabel: Light 1

 * actual behavior

NodeLabel: Light 1 should be seen in logs

 * steps to reproduce
Commission Raspberry pi with DUT as a Hub.

[raspberry-brbinfo.txt](https://github.com/project-chip/connectedhomeip/files/9239570/raspberry-brbinfo.txt)
[DUThub-brbinfo.txt](https://github.com/project-chip/connectedhomeip/files/9239579/DUThub-brbinfo.txt)

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-08-02:
@revanthb28 On the specific version of bridge-app you are running, what does
```
chip-tool descriptor read parts-list 1 0
```
return?

revanthb28, 2022-08-02:
Hi Boris,

We are trying with all-cluster-app.

bzbarsky-apple, 2022-08-03:
all-cluster-app does not have an endpoint 3, so UNSUPPORTED_ENDPOINT is correct there.

closed 2022-08-03 by bzbarsky-apple