project-chip/connectedhomeip #21539: Ensure that the device pointer we get from `getConnectedDevicePointer` is released when no longer used

tehampson, 2022-08-02:
#### Problem
With https://github.com/project-chip/connectedhomeip/pull/21256/ we are introducing a memory leak. The leak was allowed to be introduced since that PR fixes a use after free that is arguably worse then the memory leak. The PR was large enough without refactoring the java side hence creating this PR so that the memory leak could be addressed separately afterwards

 #### Proposed Solution
When onDeviceConnected gives a pointer to the Java side. We wrap the pointer in an object that has a desctructor that calls a JNI to do a free on that pointer provided. This would need all downstream consumers that previously passed a long the pointer to now use the wrapper object.

yunhanw-google, 2023-03-11:
is this still issue? @tehampson

maurycy-yolabs, 2024-03-19:
I don't understand why using something like `try { ... get and use pointer } finally { releasePointer }` is not a viable solution. Folks can create helpers if needed unless this is meant to track that work. 

@yunhanw-google this is still present within the android chiptool project.

maurycyw, 2024-03-19:
I don't understand why using something like `try { ... get and use pointer } finally { releasePointer }` is not a viable solution. (Especially since folks can create helpers if needed unless this is meant to track that work)

[@yunhanw-google](https://github.com/yunhanw-google) this is still present within the android chiptool project.

tehampson, 2024-03-25:
If I remember this issue correctly. The problem is that the `onDeviceConnected` is called back from the Matter SDK event loop thread, and it uses [this callback here](https://github.com/project-chip/connectedhomeip/blob/master/src/controller/java/AndroidCallbacks.cpp#L88-L91) to provide an `OperationalDeviceProxy`

If I recall correctly, on the android side the act of connecting and the act of doing some interaction are 2 separate paths. The result of the connection provides the pointer to the `OperationalDeviceProxy`. It was this pointer, that at the time, was not being freed properly after the interaction was completed