project-chip/connectedhomeip #21544: ExchangeHolder leaks exchanges on send failure followed by session expiration

bzbarsky-apple, 2022-08-02:
#### Problem
Consider the following sequence of events:

1. An exchange is created for a session.
2. The exchange is stored in an ExchangeHolder.
3. A message is sent on the exchange.  The send fails (which means the exchange does not close).
4. Caller does not close the exchange, because it's relying on the ExchangeHolder to do that.
5. The session is evicted.  This lands in `ExchangeContext::OnSessionReleased` which sees that we're not in `IsResponseExpected()` state and hence that it's our consumer's responsibility to close us.  So it does not call `Abort` or `Close`.
6. The exchange calls `DoClose` to do its session-eviction cleanup, which calls `OnExchangeClosing` on the holder.
7. The holder sees that neither `IsResponseExpected()` nor `IsSendExpected()` is true and just forgets its pointer to the exchange.
8. ~ExchangeHolder is called, does nothing because it's not holding anything.

As a result nothing ever drops the ref to the exchange.

This is trivially reproducible on Mac with:
```
chip-tool onoff toggle 0xffffffffffff4141 1454
````
(a group command) which has code like this:
```
        chip::app::CommandSender commandSender(mCallback, exchangeManager);
        ReturnErrorOnFailure(commandSender.AddRequestDataNoTimedCheck(commandPath, value, chip::NullOptional));

        chip::Transport::OutgoingGroupSession session(groupId, fabricIndex);
        return commandSender.SendGroupCommandRequest(chip::SessionHandle(session));
```
Note that the `session` destructor happens before the `commandSender` destructore, hence step 5 above happening before step 8.  But in general the above ordering could happen for various reasons.

 #### Proposed Solution
We need to decide whether `OnExchangeClosing` means (1) "Close or Abort has been called, and the ref that was holding has been dropped" (which is what SessionHolder seems to be assuming) or whether it means (2) "we have set our kFlagClosed and there is no more communication on this exchange" (which is what is actually implemented).

Based on a quick grep, not much implements `OnExchangeClosing`:
* `DefaultOTARequestor.h` seems to expect meaning (1) of the callback.  In particular, all it does is drop the reference, which would be the wrong thing to do for the other meaning.
* `ExchangeHolder`, as discussed above, expects meaning (1).
* `TimedHandler` kind of expects meaning (2), but in practice since it's always a responder it never ends up in a state where it's got an exchange that is not processing an incoming messagne and not waiting for a response.  So it never ends up in the codepath above, where we end up in `DoClose` without dropping a ref.

But an issue there is that `DoClose` will drop the reference to the delegate, so _won't_ be able to notify it about the exchange closing in the future....

bzbarsky-apple, 2022-08-02:
Maybe `OnExchangeClosing` should have an argument that indicates whether the exchange's outstanding ref was dropped by Close/Abort (leading to OnExchangeClosing) or whether it's still up to the delegate to release the exchange....

The other option is to change `OnSessionReleased` to always drop the ref, and then ensure that every single exchange delegate implements `OnExchangeClosing` and handles it by dropping its exchange pointer.

mrjerryjohns, 2022-08-02:
So, the holder is built to assume that `OnExchangeClosing` will only get triggered as a side-effect of a closure/abort that it itself did not trigger. The entity that triggered it should not be the application itself that is using the holder, but rather, the `ExchangeContext` itself as a side effect of completion of normal work on the EC, an exchange timeout, or session closure, etc.

What it assumes today is that the ownership of the EC would have already transferred to the stack at the time `OnExchangeClosing` is called, which isn't the case when we have an exchange with the `WillSendMessage` being true AND, the stack initiated the closure of the exchange (e.g on session release). This can manifest through a session closure without even sending a message, or, sending a message that failed (as you observed in the issue).

In that case, it should be taking on that responsibility to close out the exchange and release the reference. Unfortunately, the way things are structured, application responsibility of closing the exchange on send failure is only implementable as a reaction to receiving an error in the `SendMessage` call.

It's not possible to do so from within the implementation of `OnExchangeClosing` since that will result in a use-after-free after the stack unwinds. Part of the solution space I was exploring was the following - if we have a send failure, why don't we re-instate the `WillSendMessage` flag on the EC within ExchangeContext::SendMessage right after the failure was detected? That will put it back into a state where we're basically saying, "it's back to a mode where we intend to a send a message...again".

mrjerryjohns, 2022-08-02:
> Maybe OnExchangeClosing should have an argument that indicates whether the exchange's outstanding ref was dropped by Close/Abort (leading to OnExchangeClosing) or whether it's still up to the delegate to release the exchange....

Approaching the 'who's to release the EC' question from whether Close or Abort was called vs. not is one way to answer that, but I think it should might be better it be answered against the core current state of the EC, i.e is it waiting for a response, or is it waiting to send.

i.e if I were to implement a method called `ShouldDelegateReleaseEC`, it would be implemented using that bit of information.

That's kind of the basis of what I'm talking about above.

> The other option is to change OnSessionReleased to always drop the ref, and then ensure that every single exchange delegate that might call Close or Abort implements OnExchangeClosing and handles it by dropping its exchange pointer.

This is more invasive, and amounts to changing lots of existing call-sites.
I'd rather have an in-out argument to `OnExchangeClosing`, a bool that can be set by the callee to indicate if it wants the EC to release itself, i.e `OnExchangeClosing(bool &autoReleaseIfNeeded)`.
That would be set to false by default, so that it doesn't affect existing callers.
ExchangeHolder however, could set that to 'true' in its implementation, and that would be a signal to have it be released (if needed).

bzbarsky-apple, 2022-08-03:
> Part of the solution space I was exploring was the following - if we have a send failure, why don't we re-instate the WillSendMessage flag on the EC within ExchangeContext::SendMessage right after the failure was detected? 

That is an interesting thought...  That keeps down our state space to one of "waiting for send" or "waiting for response" or "in the middle of processing messages, then will close or wait for send or wait for response".

But you are right that this is not a solution on its own, because we could still have an exchange that is in "waiting for send" state, without the send failure bits going on here, and then the session gets evicted and we end up in the same situation as what I observed in the issue, right?  That's what's happening in #21586, in fact.

> This is more invasive, and amounts to changing lots of existing call-sites.

Yes, I am not a fan of that option.  I was just listing all the things I could think of.

> I'd rather have an in-out argument to OnExchangeClosing, a bool that can be set by the callee to indicate if it wants the EC to release itself, i.e OnExchangeClosing(bool &autoReleaseIfNeeded).

Why not just use a return value instead of a boolean outparam?  The return value would have the semantics of "do you think you are currently holding a ref to this exchange?"

The other option we discussed on Slack yesterday would be a boolean arg as I described above for "did we already release our one primordial ref", and making it ok to call Close/Abort from inside DoClose...

closed 2022-08-29 by mrjerryjohns