project-chip/connectedhomeip #21550: BLEManagerImpl.cpp crash

robertfarnum, 2022-08-02:
#### Problem
Android JNIEnv::NewByteArray crash.  Stack trace attached (BLEManagerImpl.cpp.stack.txt)

 * expected behavior
 App does not crash
 * actual behavior
 App SIGABRTs; see stack trace.
 * steps to reproduce
 Provision device on Wi-Fi
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
None currently found

[BLEManagerImpl.cpp.stack.txt](https://github.com/project-chip/connectedhomeip/files/9244060/BLEManagerImpl.cpp.stack.txt)

mrjerryjohns, 2022-08-02:
The Android CHIPTool is no longer being actively maintained by the original maintainers.

robertfarnum, 2022-08-02:
The issue is in an android platform library, not just in the CHIPTool.  Is there a test suite that could or should enable reproducing the issue?

robertfarnum, 2022-08-03:
@chrisdecenzo The actual cause of the above panic is due to an incorrect signature in the AndroidDeviceControlWrapper.cpp:

 CHIP_ERROR err = JniReferences::GetInstance().FindMethod(env, mJavaObjectRef, "onCommissioningStatusUpdate",
                                                             "(JLJAVA/LANG/STRING;I)V", &onCommissioningStatusUpdateMethod);

should be:

 CHIP_ERROR err = JniReferences::GetInstance().FindMethod(env, mJavaObjectRef, "onCommissioningStatusUpdate",
                                                             "(JLjava/lang/String;I)V", &onCommissioningStatusUpdateMethod);

robertfarnum, 2022-08-03:
Related to https://github.com/project-chip/connectedhomeip/issues/21493.

andy31415, 2022-09-15:
I believe the wrong signature was fixed in #21980

closed 2022-09-15 by andy31415