project-chip/connectedhomeip #21571: Android chip-tool crash during saving wifi credentials screen

jamesluo11, 2022-08-03:
#### Problem
Complied the master latest branch(JAVA 8), got the android chip-tool.
While chip-tool crash without any prompt during saving wifi credentials screen
<what's wrong or missing, please include any applicable:
 * expected behavior
 Pairing process succeed 
 * actual behavior
 The pairing process did NOT start,   chip-tool crash after clicking "SAVE NETWORK".
 * steps to reproduce
 Just normally follow the steps: "PROVISION CHIP DEVICE WITH WI-FI" -> "SUBMIT" -> "SAVE NETWORK". Then app crash without any prompt.
 * system configuration
 oppo android phone 
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

robertfarnum, 2022-08-03:
The logs indicate this is a duplicate of https://github.com/project-chip/connectedhomeip/issues/21493 being worked on by @chrisdecenzo

andy31415, 2022-09-15:
Closed as the dup was closed.

closed 2022-09-15 by andy31415