project-chip/connectedhomeip #21573: Test_TC_DRLK_2_9 is disabled, test does not align with the spec

woody-apple, 2022-08-03:
Filing an issue for this

_Originally posted by @woody-apple in https://github.com/project-chip/connectedhomeip/pull/21319#discussion_r936227484_

manjunath-grl, 2022-08-05:
@woody-apple  we had 3 issue(20991 , 20930 and 19990)  opened for TC-DRLK-2.9 we verified issue [20991](https://github.com/project-chip/connectedhomeip/issues/20991) and [20930](https://github.com/project-chip/connectedhomeip/issues/20930) and script is updated in [PR](https://github.com/project-chip/connectedhomeip/pull/21672) , however issue reported in [19990](https://github.com/project-chip/connectedhomeip/issues/19990)  still seems to be present( expects response of 0x03 (OCCUPIED), but received a response of 0x02(Duplicate)

manjunath-grl, 2022-09-26:
@woody-apple [19990](https://github.com/project-chip/connectedhomeip/issues/19990) issue has been fixed now, scripts updated as per testplan and spec this ticket can be closed.

closed 2023-04-02 by stale[bot]