project-chip/connectedhomeip #21578: TC-DA-1.2 - Certification Type should be 1 or 2 but I am getting 0 in step 8.3

sowmyassp, 2022-08-03:
**Issue :** 
Certification Type should be 1 or 2 but I am getting 0 in step 8.3
[image]

bzbarsky-apple, 2022-08-03:
Your spec reference is wrong.  The correct spec reference is https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_attestation/Certification_Declaration_Specification.adoc#ref_CertDecl ("6.3.1. Certification Declaration (CD) Format" in the full spec), which says:

> The certification_type field SHALL contain the type of certification for this CD, interpreted according to the following table:

* 0: used for development and test purposes
* 1: provisional - used for a device when going into certification testing, or to allow production and distribution to occur in parallel with certification (with potential software fixes yielding a higher SoftwareVersion which gets certification)
* 2: official - allocated after passing certification

So for all-clusters-app 0 is correct, because it's a test app, not a real device.  Just like for all-clusters-app a vendor id outside the `[0x0001 .. 0xFFF0]` range in the test plan is correct, because it's a test app, not a real device.

bzbarsky-apple, 2022-08-03:
@cjandhyala

closed 2022-08-03 by bzbarsky-apple