project-chip/connectedhomeip #21579: My DAC Keypair signature verification code failed. Can someone help me see what the problem is?

xufangbo, 2022-08-03:
#### error message
~~~
chip-tool: ../../examples/chip-tool/commands/haier/ValidateKeypairCommand.cpp:60: void ValidateKeypairCommand::validateByPublicKey(chip::MutableByteSpan&): 
Assertion `pubkey.ECDSA_validate_hash_signature(message_to_signature.data(), message_to_signature.size(), deviceSignature) == CHIP_NO_ERROR' failed.

~~~

 #### cpp file ：

~~~cpp

#include "ValidateKeypairCommand.h"
#include "Keypair.h"
#include <assert.h>
#include <lib/core/CHIPError.h>

using namespace ::chip::Credentials;

CHIP_ERROR ValidateKeypairCommand::Run()
{
    system("clear");

    Crypto::P256ECDSASignature signature;
    MutableByteSpan signatureByPrivateKey{ signature.Bytes(), signature.Capacity() };

    this->SignByPrivateKey(signatureByPrivateKey);

    this->ValidateByPublicKey(signatureByPrivateKey);

    return CHIP_NO_ERROR;
}

void ValidateKeypairCommand::SignByPrivateKey(MutableByteSpan & signatureByPrivateKey)
{
    // message_to_sign
    uint8_t md[Crypto::kSHA256_Hash_Length];
    MutableByteSpan message_to_sign(md);
    MessageToSignature(message_to_sign);

    // keypair
    Crypto::P256Keypair keypair;
    assert(LoadKeypairFromRaw(Haier::Keypair::kDacPriKey, Haier::Keypair::kDacPubKey, keypair) == CHIP_NO_ERROR);

    // signature
    Crypto::P256ECDSASignature signature;
    assert(keypair.ECDSA_sign_msg(message_to_sign.data(), message_to_sign.size(), signature) == CHIP_NO_ERROR);

    assert(CopySpanToMutableSpan(ByteSpan{ signature.ConstBytes(), signature.Length() }, signatureByPrivateKey) == CHIP_NO_ERROR);

    assert(signatureByPrivateKey.size() == Crypto::P256ECDSASignature::Capacity());
}

void ValidateKeypairCommand::ValidateByPublicKey(MutableByteSpan & signatureByPrivateKey)
{
    // public key
    P256PublicKey pubkey;
    assert(ExtractPubkeyFromX509Cert(Haier::Keypair::kDacCert, pubkey) == CHIP_NO_ERROR);

    // signatureByPrivateKey - > deviceSignature
    P256ECDSASignature deviceSignature;
    assert(deviceSignature.SetLength(signatureByPrivateKey.size()) == CHIP_NO_ERROR);
    memcpy(deviceSignature.Bytes(), signatureByPrivateKey.data(), signatureByPrivateKey.size());

    // message_to_signature
    uint8_t md[kSHA256_Hash_Length];
    MutableByteSpan message_to_signature(md);
    MessageToSignature(message_to_signature);

    // validate
    // error here : Assertion `pubkey.ECDSA_validate_hash_signature(message_to_signature.data(), message_to_signature.size(),
    // deviceSignature) == CHIP_NO_ERROR' failed.
    assert(pubkey.ECDSA_validate_hash_signature(message_to_signature.data(), message_to_signature.size(), deviceSignature) ==
           CHIP_NO_ERROR);
}

CHIP_ERROR LoadKeypairFromRaw(ByteSpan private_key, ByteSpan public_key, Crypto::P256Keypair & keypair)
{
    Crypto::P256SerializedKeypair serialized_keypair;
    ReturnErrorOnFailure(serialized_keypair.SetLength(private_key.size() + public_key.size()));
    memcpy(serialized_keypair.Bytes(), public_key.data(), public_key.size());
    memcpy(serialized_keypair.Bytes() + public_key.size(), private_key.data(), private_key.size());
    return keypair.Deserialize(serialized_keypair);
}

void MessageToSignature(MutableByteSpan & messageDigestSpan)
{
    Hash_SHA256_stream hashStream;
    assert(hashStream.Begin() == CHIP_NO_ERROR);
    assert(hashStream.AddData(Haier::Keypair::kAttestationElements) == CHIP_NO_ERROR);
    assert(hashStream.AddData(Haier::Keypair::kAttestationChallenge) == CHIP_NO_ERROR);
    assert(hashStream.Finish(messageDigestSpan) == CHIP_NO_ERROR);
}

~~~

thanks

bzbarsky-apple, 2022-08-03:
Have you considered logging what the actual error is?  Something like:
```
    CHIP_ERROR err = pubkey.ECDSA_validate_hash_signature(message_to_signature.data(), message_to_signature.size(), deviceSignature);
    if (err != CHIP_NO_ERROR) {
        fprintf(stderr, "Failed signature validation: %" CHIP_ERROR_FORMAT, err.Format());
   }
```

On Mac and Linux by default that will tell you where the error was created, so you can see where inside `ECDSA_validate_hash_signature` things failed....

xufangbo, 2022-08-09:
@bzbarsky-apple thanks,use you suggest code ,the error is 
~~~
Failed signature validation: ../../src/crypto/CHIPCryptoPALOpenSSL.cpp:737: Error 0x00000014
~~~

bzbarsky-apple, 2022-08-09:
@xufangbo And what is on line 737 in your version of `src/crypto/CHIPCryptoPALOpenSSL.cpp`?  Or what SHA are you testing on?

That said, I would expect this is this bit of code:
```
    result = ECDSA_do_verify(Uint8::to_const_uchar(hash), static_cast<boringssl_size_t_openssl_int>(hash_length), ec_sig, ec_key);
    VerifyOrExit(result == 1, error = CHIP_ERROR_INVALID_SIGNATURE);
```
since 0x00000014 is CHIP_ERROR_INVALID_SIGNATURE.

Which means the signature involved is in fact wrong, right?

xufangbo, 2022-08-19:
Thank you very much for your enthusiasm and patience。

It's 737 lines of code, but I can't understand ECDSA_ do_ Verify the code in
I use matter's chip::DevelopmentCerts::kDevelopmentDAC_Cert_FFF1_8000 related certificates and keypair。
But mesge_ to_ sign is my own construction

xufangbo, 2022-08-22:
@bzbarsky-apple thanks ,I have fixed my mistake
change ECDSA_validate_hash_signature() to ECDSA_validate_msg_signature()

closed 2022-08-22 by xufangbo