project-chip/connectedhomeip #21586: [crash] Crash on fail-safe timeout

Damian-Nordic, 2022-08-03:
#### Problem
Was testing the commissioning using incorrect network credentials and the following crash happened during the fail-safe timeout:
```
#0  _exit (status=<optimized out>) at /home/damian/src/ncs/zephyr/lib/libc/newlib/libc-hooks.c:281
#1  0x000be9da in abort ()
#2  0x000aaf98 in chipAbort () at ../../../../../../src/lib/support/CodeUtils.h:506
#3  chip::Messaging::ExchangeHolder::operator-> (this=<optimized out>, this=<optimized out>) at ../../../../../../src/messaging/ExchangeHolder.h:111
#4  0x000623b2 in chip::app::CommandHandler::DecrementHoldOff (this=0x2000d0cc <chip::app::sInteractionModelEngine+52>) at ../../../../../../src/app/CommandHandler.cpp:189
#5  chip::app::CommandHandler::DecrementHoldOff (this=0x2000d0cc <chip::app::sInteractionModelEngine+52>) at ../../../../../../src/app/CommandHandler.cpp:179
#6  0x000ab3ea in chip::app::CommandHandler::Handle::Release (this=0x200127bc <chip::DeviceLayer::Internal::(anonymous namespace)::sThreadNetworkCommissioningInstance+68>)
    at ../../../../../../src/app/CommandHandler.cpp:585
#7  chip::app::CommandHandler::Handle::Release (this=0x200127bc <chip::DeviceLayer::Internal::(anonymous namespace)::sThreadNetworkCommissioningInstance+68>)
    at ../../../../../../src/app/CommandHandler.cpp:579
#8  0x000b4f0a in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::DispatchEventToApplication (
    this=this@entry=0x20010878 <chip::DeviceLayer::PlatformManagerImpl::sInstance+8>, event=event@entry=0x2002b778 <chip::DeviceLayer::sChipThreadStack+8184>)
    at ../../../../../../src/include/platform/internal/GenericPlatformManagerImpl.ipp:298
#9  0x0006db40 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_DispatchEvent (
    this=this@entry=0x20010878 <chip::DeviceLayer::PlatformManagerImpl::sInstance+8>, event=event@entry=0x2002b778 <chip::DeviceLayer::sChipThreadStack+8184>)
    at ../../../../../../src/include/platform/internal/GenericPlatformManagerImpl.ipp:263
#10 0x000b4f86 in chip::DeviceLayer::PlatformManager::DispatchEvent (event=0x2002b778 <chip::DeviceLayer::sChipThreadStack+8184>, 
    this=0x20010870 <chip::DeviceLayer::PlatformManagerImpl::sInstance>) at ../../../../../../src/include/platform/PlatformManager.h:437
```

100% reproducible on a494b1c396d1c97d912ccdc707774c4cc27a6125 - `VerifyOrDie(mpExchangeCtx != nullptr)` assertion fails.
Tested on nRF Connect Lock example, but looks like a generic problem.

 #### Proposed Solution
N/A

Damian-Nordic, 2022-08-03:
@mrjerryjohns Looks like 251d49fcc42ead8694cae48b14fdbdc9d049363f has introduced this issue. Apparently, `CommandHandler::DecrementHoldOff` assumes that the exchange is not closed.

bzbarsky-apple, 2022-08-03:
In this case the exchange is not in fact closed, in the sense that its last ref never dropped.  The difference is that fail-safe timeout kills the PASE session, which puts us in the state described in https://github.com/project-chip/connectedhomeip/issues/21544: we're in "will send message" state (and in particular not "response expected" state), hence we `DoClose` without dropping the ref.  The thing #21081 introduced here is nulling out the exchange pointer even though we are still holding a ref to it.  That means we leak the exchange _and_ end up with this null-deref....

bzbarsky-apple, 2022-08-03:
So to be clear, in addition to the fix for #21544 we would need to add the requisite null-checks here.  And we need to check the other ExchangeHolder users to make sure they are not missing any null-checks, since now the exchange can become null in ways it could not before.

Damian-Nordic, 2022-08-03:
Added a fix for this issue as it blocks me with another task, but I agree other users need to be reviewed as well: https://github.com/project-chip/connectedhomeip/pull/21590.

PR #21590 (2022-08-08): Fix crash on fail-safe timeout due to closed exchange
#### Problem
When CommandHandler is released it dereferences an exchange holder to figure out if the status response can be sent.
The exchange holder, however, can be nulled-out at this point.

#### Change overview
Add a null check. 
Fixes #21586

#### Testing
Added a unit test that reproduces the crash scenario.
closed 2022-08-08 by Damian-Nordic