project-chip/connectedhomeip #21625: Actions cluster, enum ActionStruct, field State is printed as "Status" in chip-tool

leorozendaal, 2022-08-04:
#### Problem
Actions cluster, enum ActionStruct, field State is printed as "Status" in chip-tool, see example below

 #### Proposed Solution
Rename "Status" to "State" (at least in the printed output, consider to rename the variable name internally) to match the spec

Example of chip-tool output:
```
[1658473018.104129][16539:16544] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0025 Attribute 0x0000_0000 DataVersion: 3292516604
[1658473018.104215][16539:16544] CHIP:TOO:   action list: 1 entries
[1658473018.104263][16539:16544] CHIP:TOO:     [1]: {
[1658473018.104728][16539:16544] CHIP:TOO:       ActionID: 4097
[1658473018.104753][16539:16544] CHIP:TOO:       Name: Room 1 On
[1658473018.104770][16539:16544] CHIP:TOO:       Type: 3
[1658473018.104786][16539:16544] CHIP:TOO:       EndpointListID: 57345
[1658473018.104803][16539:16544] CHIP:TOO:       SupportedCommands: 1
[1658473018.104819][16539:16544] CHIP:TOO:       Status: 0
[1658473018.104834][16539:16544] CHIP:TOO:      }

```

bzbarsky-apple, 2022-08-04:
chip-tool just uses whatever is in the XML.  The problem is that the XML does not match the spec.

I am actually in the middle of fixing that last for this cluster; it had a bunch of other issues too.  Taking.

leorozendaal, 2022-08-04:
@bzbarsky-apple  thanks!

bzbarsky-apple, 2022-08-05:
@leorozendaal No problem!

closed 2022-08-08 by bzbarsky-apple