project-chip/connectedhomeip #21626: Bind back to yourself -- loopback binding

jonsmirl, 2022-08-04:
Complex devices need to be able to do bindings between endpoints on the same device. I tried doing this and discovery is unable to find its own address. How should this be handled? 

accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null },{"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": [29315], "targets": null }]' 0x7283 0

binding write binding '[{"fabricIndex": 1, "node":29315, "endpoint":1, "cluster":6}]' 0x7283 0x2

But it can't establish a connection to itself....
I (207833) chip[DIS]: Checking node lookup status after 15000 ms
E (207833) chip[DIS]: Operational discovery failed for 0x0000000000007283: 32
E (207843) chip[SVR]: Failed to establish connection to node 0x0000000000007283

Any ideas on how to establish a loopback (self-bound) connection?

I have not tried this, but this also impacts bridges. Will the bridge be able to loop back devices?

jonsmirl, 2022-08-04:
Some other things to consider...

What about groups? When the device is in the group, and when it is not in the group.
if there is a self-binding and then a network unicast binding is made, should it replace or override the self binding?
If all bindings are removed should the device revert to being self bound? ie override vs replace
How can we ship the device pre-configured in a self-bound state?
Consider a smart light switch. It is an integrated switch+bulb. It should ship in the self-bound state. But then the customer may choose to add it to a group -- like a synchronized pair at top and bottom of stair case. Then the customer removes it from the group and expects it to revert to self-bound mode.

jonsmirl, 2022-08-04:
Rules like this could work:

Allow the device to create a default self-bind without needing to be on a fabric. This self bind is permanent.
If you add a network unicast bind this bind will override the default bind until removed.
If bound to a group, send the multicast on the network -- and use the default self-bind.
3a) more complex - check if the device is in the group or not, use self bind if it is.

[image]

jonsmirl, 2022-08-04:
Should Matter have special loop back node and group IDs? for example 0xFFFF always means self?

closed 2023-02-12 by stale[bot]

LongND0105, 2023-08-08:
> I have not tried this, but this also impacts bridges. Will the bridge be able to loop back devices?

@jonsmirl Have you tried bind a matter endpoint in the bridge to the another device. I'm trying to here: #28557. Could you give me some ideas?