project-chip/connectedhomeip #21627: Need YAML support for validating constraint or null

manjunath-grl, 2022-08-04:
Summary:
Need support for test steps where we need to check if a values is null, if the values is not null we need to validate its constraints
Effected tests:
TC-RH-2.1
TC-TMP-2.1
TC-FLW-2.1
Testplan screenshot:
[image]

bzbarsky-apple, 2022-08-04:
@manjunath-grl I don't understand what the issue is.  For a nullable value min/max constraints always pass if the value is null.

That is, if you write this in the YAML:
```
  constraints:
      minValue: 1
      maxValue: 0xfffd
```
then the behavior as a function of the value will be:

* null: pass
* non-null, less than 1: fail
* non-null, greater than 0xfffd: fail
* non-null, otherwise: pass

closed 2022-08-04 by bzbarsky-apple

manjunath-grl, 2022-08-04:
@bzbarsky-apple thanks, we didn't know that constraints checks cover null values as well.