project-chip/connectedhomeip #21628: Compressed fabric ID is: 0x0000000000000000

HubertHugo, 2022-08-04:
SHA: 41cb8220744f2587413d0723e24847f07d6ac59f

When commissioning I get logs:
```
[...]
[1659615339.789566][4785:4785] CHIP:FP: Added new fabric at index: 0x1
[1659615339.789569][4785:4785] CHIP:FP: Assigned compressed fabric ID: 0x0AF1E75340BFF486, node ID: 0x000000000001B669
[1659615339.789572][4785:4785] CHIP:TS: Last Known Good Time: 2022-08-04T06:10:48
[1659615339.789574][4785:4785] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1659615339.789575][4785:4785] CHIP:TS: Retaining current Last Known Good Time
[1659615339.789762][4785:4785] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1659615339.790040][4785:4785] CHIP:TS: Committing Last Known Good Time to storage: 2022-08-04T06:10:48
[1659615339.790324][4785:4785] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1659615339.790333][4785:4785] CHIP:IN: UDP::Init bind&listen port=5550
[1659615339.790345][4785:4785] CHIP:IN: UDP::Init bound to port=5550
[1659615339.790348][4785:4785] CHIP:IN: UDP::Init bind&listen port=5550
[1659615339.790357][4785:4785] CHIP:IN: UDP::Init bound to port=5550
[1659615339.790359][4785:4785] CHIP:IN: TransportMgr initialized
[1659615339.790374][4785:4785] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1659615339.790377][4785:4785] CHIP:CTL: Setting attestation nonce to random value
[1659615339.790382][4785:4785] CHIP:CTL: Setting CSR nonce to random value
[1659615339.790668][4785:4785] CHIP:CTL: Generating NOC
[1659615339.790854][4785:4785] CHIP:FP: Validating NOC chain
[1659615339.791232][4785:4785] CHIP:FP: NOC chain validation successful

[1659615339.791282][4785:4785] CHIP:FP: Added new fabric at index: 0x2
[1659615339.791286][4785:4785] CHIP:FP: Assigned compressed fabric ID: 0x4DC309644D2EEE22, node ID: 0x000000000001B669
[1659615339.791289][4785:4785] CHIP:TS: Last Known Good Time: 2022-08-04T06:10:48
[1659615339.791292][4785:4785] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1659615339.791293][4785:4785] CHIP:TS: Retaining current Last Known Good Time
[1659615339.791496][4785:4785] CHIP:FP: Metadata for Fabric 0x2 persisted to storage.
[1659615339.791801][4785:4785] CHIP:TS: Committing Last Known Good Time to storage: 2022-08-04T06:10:48
[1659615339.792080][4785:4785] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x0000000000000000
[1659615339.792087][4785:4785] CHIP:IN: UDP::Init bind&listen port=5550
[1659615339.792098][4785:4785] CHIP:IN: UDP::Init bound to port=5550
[1659615339.792100][4785:4785] CHIP:IN: UDP::Init bind&listen port=5550
[1659615339.792106][4785:4785] CHIP:IN: UDP::Init bound to port=5550
[1659615339.792108][4785:4785] CHIP:IN: TransportMgr initialized
[1659615339.792122][4785:4785] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1659615339.792125][4785:4785] CHIP:CTL: Setting attestation nonce to random value
[1659615339.792131][4785:4785] CHIP:CTL: Setting CSR nonce to random value
[1659615339.792401][4785:4785] CHIP:CTL: Generating NOC
[1659615339.792590][4785:4785] CHIP:FP: Validating NOC chain
[1659615339.792962][4785:4785] CHIP:FP: NOC chain validation successful

[1659615339.793014][4785:4785] CHIP:FP: Added new fabric at index: 0x3
[1659615339.793017][4785:4785] CHIP:FP: Assigned compressed fabric ID: 0xC172956F12824155, node ID: 0x000000000001B669
[1659615339.793021][4785:4785] CHIP:TS: Last Known Good Time: 2022-08-04T06:10:48
[1659615339.793023][4785:4785] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1659615339.793025][4785:4785] CHIP:TS: Retaining current Last Known Good Time
[1659615339.793239][4785:4785] CHIP:FP: Metadata for Fabric 0x3 persisted to storage.
[1659615339.793555][4785:4785] CHIP:TS: Committing Last Known Good Time to storage: 2022-08-04T06:10:48
[1659615339.793851][4785:4785] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0x0000000000000000
[...]
```
so there is line 
```Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000```
It becomes a problem when commissioning second commissioner since from this output:
```
[...]
[1659615352.057490][4792:4792] CHIP:FP: Updated fabric at index: 0x2, Node ID: 0x000000000001B669
[1659615352.057494][4792:4792] CHIP:TS: Last Known Good Time: 2022-08-04T06:10:48
[1659615352.057496][4792:4792] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1659615352.057497][4792:4792] CHIP:TS: Retaining current Last Known Good Time
[1659615352.057857][4792:4792] CHIP:FP: Metadata for Fabric 0x2 persisted to storage.
[1659615352.058207][4792:4792] CHIP:TS: Committing Last Known Good Time to storage: 2022-08-04T06:10:48
[1659615352.058532][4792:4792] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x0000000000000000
[1659615352.058539][4792:4792] CHIP:IN: UDP::Init bind&listen port=5550
[1659615352.058550][4792:4792] CHIP:IN: UDP::Init bound to port=5550
[1659615352.058555][4792:4792] CHIP:IN: UDP::Init bind&listen port=5550
[1659615352.058562][4792:4792] CHIP:IN: UDP::Init bound to port=5550
[1659615352.058563][4792:4792] CHIP:IN: TransportMgr initialized
[1659615352.058579][4792:4792] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1659615352.058582][4792:4792] CHIP:CTL: Setting attestation nonce to random value
[1659615352.058588][4792:4792] CHIP:CTL: Setting CSR nonce to random value
[1659615352.058800][4792:4792] CHIP:CTL: Generating NOC
[1659615352.059088][4792:4792] CHIP:FP: Validating NOC chain
[1659615352.059462][4792:4792] CHIP:FP: NOC chain validation successful
[...]
```
I can't get fabric ID, only first commissioner has information about fabric ID for first 3 commissioners.

HubertHugo, 2022-08-05:
This first log is when first commissioner commissions (adds these 3 fabrics). The other one is when commissioning second commissioner.

I need it for testing that fabric ID from one controller on different DUTs in multi fabric scenario is equal.

bzbarsky-apple, 2022-08-05:
> I need it for testing that fabric ID from one controller on different DUTs in multi fabric scenario is equal.

I'm really not following this.  What is "one controller on different DUTs" here?

HubertHugo, 2022-08-08:
1. Commission to DUT1 as first commissioner
2. Commission to DUT2 as second/third commissioner
3. Verify that fabric IDs are the same

bzbarsky-apple, 2022-08-08:
> Verify that fabric IDs are the same

Is the commissioner in this case `chip-tool`?  What exact chip-tool commands are being run?

HubertHugo, 2022-08-17:
Yes commissioner is chip-tool.
Commands used:
```
pairing ble-thread 1 hex:[...] 20202021 3584 --commissioner-name alpha
pairing ble-thread 2 hex:[...] 20202021 3584 --commissioner-name beta  (it's connecting to another DUT)
administratorcommissioning open-basic-commissioning-window 500 1 0 --timedInteractionTimeoutMs 500 --commissioner-name alpha
pairing onnetwork 1 20202021 --commissioner-name beta
```
now compare that fabric IDs printed on pairing were the same on commissioner beta

bzbarsky-apple, 2022-08-26:
> now compare that fabric IDs printed on pairing were the same on commissioner beta

OK.  I'm not sure why you would want to do that, but ok.  PR #21297 will fix this once it actually merges...

closed 2022-11-03 by tcarmelveilleux