project-chip/connectedhomeip #21630: [ESP32] Commissioning fails and watchdog not reset when pairing with wrong passcode

innovation-matters-iot, 2022-08-04:
#### Problem
when using a factory_bin prepared as described [here](https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/esp32/README.md) (with flash encryption, not tested without) and the pairing process is started (ble), with a different passcode than the one used during the creation of the factory bin, the commissioning process gets aborted in chip-tool and the DUT needs to be rebooted
[stuck_with_wrong_passcode.txt](https://github.com/project-chip/connectedhomeip/files/9259919/stuck_with_wrong_passcode.txt)

dhrishi, 2022-08-04:
@shubhamdp PTAL

shubhamdp, 2022-08-10:
I tried on 608b143300dd46ab3948ade6dcdbb40aa4206cdb with all-clusters-app and I was able to recommission the device again. Could you share the complete log capture?

innovation-matters-iot, 2022-08-10:
I just set everything up for the all-clusters-app and was also not able to reproduce. Seems to be a DUT problem on my side.

dhrishi, 2022-08-12:
Ok. Closing this now. Please feel free to reopen if you recreate it by any chance.

closed 2022-08-12 by dhrishi