project-chip/connectedhomeip #21633: Should Matter have a special address for loopback?

jonsmirl, 2022-08-04:
Should a node address of 0x0 or -1 always indicate loopback?

jonsmirl, 2022-08-04:
Related issue, could a fabric ID of 0x0 or -1 always indicate a device local fabric?  ie a fabric which is only valid internally to the device and doesn't need encryption.

bzbarsky-apple, 2022-08-04:
Currently in the spec:
* `0x0000_0000_0000_0000` is reserved as unspecified node id.
* `0xFFFF_FFFF_FFFF_FFFF` is the "all nodes" group node id.

There are some reserved group id ranges that could be used for "loopback" in a following spec revision, though.

Similarly, fabric id 0 is already reserved as "SHALL NOT be used as the identifier of a fabric."  There are no other reserved values for fabric id.

jonsmirl, 2022-08-04:
This topic needs discussion...

1) How can a simple smart light switch ship with its Switch pre-bound to the Light? A device internal, always commissioned fabric allows the light switch to have a "default' binding. But there may be other ways to achieve the same goal.

2) How should controller added unicast bindings interact with this 'default' binding? I would expect a controlled added unicast binding to override the 'default' binding. But then if the controller removes that unicast binding, the 'default' binding should reassert. 

3) What happens if two different external controllers set up conflicting unicast bindings? Does this effectively become a group binding? Should an action on one fabric trigger a response on both fabrics?

4) How should group bindings interact with the 'default' binding? Two choices: groups never loop back, so both the group and the 'default' binding are active.  Or: groups allow self-membership which can loop back.  When the group is removed, the 'default' binding should reassert.

closed 2023-10-15 by stale[bot]