project-chip/connectedhomeip #21634: [TC-SWTCH-2.2] - switch position should not go beyond the position configured for Dut[Raspberrypi]

somu1710, 2022-08-04:
**Description:** Current switch position should not go beyond the max number of switch positions configured. If a tester tries to set the currentposition to greater than Max Number of positions, DUT should reject and not set that value. 

In the current SDK development, Max number of positions are configured as 2.If the DUT supports more number of positions,It should increase to the next level.

**Actual outcome:** 
   If user try to set the currentposition as greater than or equal to max numberofpositions, DUT accepts and set the value. While reading the currentposition, it gives the updated value. 

**Expected outcome:**   
      If user try to set the currentposition as greater than or equal to max numberofpositions, DUT should reject and not set the value. While reading the currentposition, it should respond with maxnumberofposition - 1 value.

```
DUT - Raspi
App - All-clusters-app
Platform - Linux
```

**Spec link:**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Switch.adoc#features

**Testplan link:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc#tc-swtch-2-2-primary-functionality-with-server-as-dut

**Logs for reference:**

[TC-SWTCH-2.2 DUT.txt](https://github.com/project-chip/connectedhomeip/files/9260144/TC-SWTCH-2.2.DUT.txt)
[TC-SWTCH-2.2 CTRL.txt](https://github.com/project-chip/connectedhomeip/files/9260146/TC-SWTCH-2.2.CTRL.txt)
[TC-SWTCH-2.2 EVENT.txt](https://github.com/project-chip/connectedhomeip/files/9260147/TC-SWTCH-2.2.EVENT.txt)

bzbarsky-apple, 2022-08-04:
Is this just about the behavior of the example server apps?  Why do we care about what behavior those have on invalid inputs in this case?

As in, as far as I can tell this only affects:
1. Verifying that the test plans are actually reasonable, not certification of actual devices.
2. Incorrect use of the test example app in the process.

It does not seem different from running the wrong example app, or running the right example app but triggering the wrong events on it, etc, etc.

leorozendaal, 2022-08-04:
@somu1710   Maybe I don't understand the test setup.
A real product would never set a value outside the range 0..NumPositions (and if it would, it would fail TC-SWTCH-2.1 step 2b)
With your simulated DUT, which you "feed" a new position, you have to provide a sensible value (e.g. 0 or 1 not 3 in case NumPositions=2) - I've made similar fixes to verification steps sheet for SWTCH today.

bzbarsky-apple, 2022-08-04:
I _think_ this a request for our test app to catch invalid inputs here and provide better feedback to testers so they will not accidentally mess things up.  Which is a totally sensible thing to do, but might not be as high-priority as other things right now....

closed 2023-10-15 by stale[bot]