project-chip/connectedhomeip #21641: [TC-ACE-1.2] Subscriptions,  Expected Outcome is Unsupported Access(0X7e) but TH2 receives a Subscription report from the DUT .

aswathygrl, 2022-08-04:
**Test Case ID**

 [TC-ACE-1.2] Subscriptions
**Describe Issue:**
Step 18 , Step 19 :  Expected Outcome is Unsupported Access(0X7e) but TH2 receives a Subscription report from the DUT .

- TH2 Subscribes to all test attributes and events expecting Unsupported Access  for privileges less than 5.

- Step 24, Step 29,Step 33 :  (Steps 18 and 19 are Repeating )

bzbarsky-apple, 2022-08-04:
I just tried this using chip-tool interactive mode:
```
accesscontrol write acl '[{"privilege": 4, "authMode": 2, "subjects":[112233], "targets":null}]' 17 0 

operationalcredentials subscribe nocs 2 10 17 0  
```

During the priming read I get:
```
[1659635146155] [78210:5525239] CHIP: [DMG] ReportDataMessage =
[1659635146155] [78210:5525239] CHIP: [DMG] {
[1659635146155] [78210:5525239] CHIP: [DMG] 	SubscriptionId = 0x1497d9ab,
[1659635146155] [78210:5525239] CHIP: [DMG] 	AttributeReportIBs =
[1659635146155] [78210:5525239] CHIP: [DMG] 	[
[1659635146156] [78210:5525239] CHIP: [DMG] 		AttributeReportIB =
[1659635146156] [78210:5525239] CHIP: [DMG] 		{
[1659635146156] [78210:5525239] CHIP: [DMG] 			AttributeStatusIB =
[1659635146156] [78210:5525239] CHIP: [DMG] 			{
[1659635146156] [78210:5525239] CHIP: [DMG] 				AttributePathIB =
[1659635146156] [78210:5525239] CHIP: [DMG] 				{
[1659635146156] [78210:5525239] CHIP: [DMG] 					Endpoint = 0x0,
[1659635146156] [78210:5525239] CHIP: [DMG] 					Cluster = 0x3e,
[1659635146156] [78210:5525239] CHIP: [DMG] 					Attribute = 0x0000_0000,
[1659635146156] [78210:5525239] CHIP: [DMG] 				}
[1659635146156] [78210:5525239] CHIP: [DMG] 					
[1659635146156] [78210:5525239] CHIP: [DMG] 				StatusIB =
[1659635146156] [78210:5525239] CHIP: [DMG] 				{
[1659635146156] [78210:5525239] CHIP: [DMG] 					status = 0x7e (UNSUPPORTED_ACCESS),
[1659635146156] [78210:5525239] CHIP: [DMG] 				},
[1659635146156] [78210:5525239] CHIP: [DMG] 				
[1659635146156] [78210:5525239] CHIP: [DMG] 			},
[1659635146156] [78210:5525239] CHIP: [DMG] 			
[1659635146156] [78210:5525239] CHIP: [DMG] 		},
[1659635146156] [78210:5525239] CHIP: [DMG] 		
[1659635146156] [78210:5525239] CHIP: [DMG] 	],
[1659635146156] [78210:5525239] CHIP: [DMG] 	
[1659635146156] [78210:5525239] CHIP: [DMG] 	InteractionModelRevision = 1
[1659635146156] [78210:5525239] CHIP: [DMG] }
[1659635146156] [78210:5525239] CHIP: [TOO] Response Failure: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)
```

@aswathygrl   What exact results are you seeing?  What exact commands are you running?

aswathygrl, 2022-08-05:
commit f98f73380c22f410ff6f323f182774a78154b1ce (HEAD -> main, origin/main)  , I had run below test steps in Interactive mode. logs are attached from Step 16 to 19.
[TC-ACE - 1.2 ] Issues  IN STEPS 18 AND 19 .txt](https://github.com/project-chip/connectedhomeip/files/9266310/TC-ACE.-.SDK.ISSUE.IN.STEPS.18.AND.19.txt)

bzbarsky-apple, 2022-08-05:
@aswathygrl Looking at that log, the commands are doing the following:

1. Set ACLs as follows:
    a. Allow node 112233 to Administer endpoint 0.
    b. Allow node 112233 to Manage endpoint 0.
2. Do various things with endpoint 0.

Since node 112233 is explicitly allowed to both Manage and Administer EP0, it can do all the "various things".

Just to be clear, an action is allowed if there is _any_ ACL in the list allowing the action.

bzbarsky-apple, 2022-08-05:
@cjandhyala The verification steps here need fixing.

closed 2022-08-05 by bzbarsky-apple

lisaaaas, 2023-01-17:
I would like to know why instruction ./chip-tool interactive start operationalcredentials subscribe nocs  1 30 2 0 --fabric-filtered false  --keepSubscriptions 0 in the ACE-1.2 use case appears timeout after execution. Does TH2 mean to restart a new pc？

bzbarsky-apple, 2023-01-17:
@lisaaaas `./chip-tool interactive start operationalcredentials subscribe nocs 1 30 2 0 --fabric-filtered false --keepSubscriptions 0` is not a valid command.  What actual commands are being run?

lisaaaas, 2023-01-18:
I am testing the ACE-1.2, first with networking, then with instruction accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0， then with instruction  ./chip-tool interactive start operationalcredentials subscribe nocs  1 30 2 0 --fabric-filtered false  --keepSubscriptions 0 with timeout.

lisaaaas, 2023-01-18:
`Pls run this test in chip tool interactive mode  using the command  ./chip-tool interactive start 

accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": [{ "cluster": null, "endpoint": 0, "deviceType": null }]}]' 1 0
On TH(Chiptool)1,Verify successfully  ACL attribute with a list of AccessControlEntryStruct entries containing 1 element.
[1659508250.468539][10228:10233] CHIP:EM: Removed CHIP MessageCounter:188603873 from RetransTable on exchange 14900i
[1659508250.468582][10228:10233] CHIP:DMG: WriteClient moving to [ResponseRe]
[1659508250.468646][10228:10233] CHIP:DMG: WriteResponseMessage =
[1659508250.468679][10228:10233] CHIP:DMG: {
[1659508250.468706][10228:10233] CHIP:DMG:      AttributeStatusIBs =
[1659508250.468745][10228:10233] CHIP:DMG:      [
[1659508250.468775][10228:10233] CHIP:DMG:              AttributeStatusIB =
[1659508250.468809][10228:10233] CHIP:DMG:              {
[1659508250.468841][10228:10233] CHIP:DMG:                      AttributePathIB =
[1659508250.468881][10228:10233] CHIP:DMG:                      {
[1659508250.468922][10228:10233] CHIP:DMG:                              Endpoint = 0x0,
[1659508250.468964][10228:10233] CHIP:DMG:                              Cluster = 0x1f,
[1659508250.469005][10228:10233] CHIP:DMG:                              Attribute = 0x0000_0000,
[1659508250.469044][10228:10233] CHIP:DMG:                      }
[1659508250.469087][10228:10233] CHIP:DMG:
[1659508250.469124][10228:10233] CHIP:DMG:                      StatusIB =
[1659508250.469163][10228:10233] CHIP:DMG:                      {
[1659508250.469202][10228:10233] CHIP:DMG:                              status = 0x00 (SUCCESS),
[1659508250.469240][10228:10233] CHIP:DMG:                      },
[1659508250.469279][10228:10233] CHIP:DMG:
[1659508250.469310][10228:10233] CHIP:DMG:              },
[1659508250.469354][10228:10233] CHIP:DMG:
[1659508250.469384][10228:10233] CHIP:DMG:              AttributeStatusIB =
[1659508250.469417][10228:10233] CHIP:DMG:              {
[1659508250.469448][10228:10233] CHIP:DMG:                      AttributePathIB =
[1659508250.469485][10228:10233] CHIP:DMG:                      {
[1659508250.469523][10228:10233] CHIP:DMG:                              Endpoint = 0x0,
[1659508250.469564][10228:10233] CHIP:DMG:                              Cluster = 0x1f,
[1659508250.469604][10228:10233] CHIP:DMG:                              Attribute = 0x0000_0000,
[1659508250.469644][10228:10233] CHIP:DMG:                              ListIndex = Null,
[1659508250.469682][10228:10233] CHIP:DMG:                      }
[1659508250.469723][10228:10233] CHIP:DMG:
[1659508250.469759][10228:10233] CHIP:DMG:                      StatusIB =
[1659508250.469796][10228:10233] CHIP:DMG:                      {
[1659508250.469834][10228:10233] CHIP:DMG:                              status = 0x00 (SUCCESS),
[1659508250.469873][10228:10233] CHIP:DMG:                      },
[1659508250.469912][10228:10233] CHIP:DMG:
[1659508250.469943][10228:10233] CHIP:DMG:              },
[1659508250.469979][10228:10233] CHIP:DMG:
[1659508250.470007][10228:10233] CHIP:DMG:      ],
[1659508250.470048][10228:10233] CHIP:DMG:
[1659508250.470077][10228:10233] CHIP:DMG:      InteractionModelRevision = 1
[1659508250.470105][10228:10233] CHIP:DMG: }
[1659508250.470228][10228:10233] CHIP:DMG: WriteClient moving to [AwaitingDe]
[1659508250.470299][10228:10233] CHIP:EM: Sending Standalone Ack for MessageCounter:48929405 on exchange 14900i
[1659508250.470380][10228:10233] CHIP:IN: Prepared secure message 0xffff994999e8 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 14900i with MessageCounter:188603874.
[1659508250.470426][10228:10233] CHIP:IN: Sending encrypted msg 0xffff994999e8 with MessageCounter:188603874 to 0x0000000000000001 (1) at monotonic time: 000000000598DDBE msec
[1659508250.470575][10228:10233] CHIP:EM: Flushed pending ack for MessageCounter:48929405 on exchange 14900i
[1659508250.470779][10228:10228] CHIP:CTL: Shutting down the commissioner
[1659508250.470818][10228:10228] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1659508250.470842][10228:10228] CHIP:CTL: Shutting down the controller
[1659508250.470874][10228:10228] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1659508250.470904][10228:10228] CHIP:IN: SecureSession[0xffff84001930]: MarkForEviction Type:2 LSID:16459
[1659508250.470931][10228:10228] CHIP:SC: SecureSession[0xffff84001930]: Moving from state 'kActive' --> 'kPendingEviction'
[1659508250.470959][10228:10228] CHIP:IN: SecureSession[0xffff84001930]: Released - Type:2 LSID:16459
[1659508250.470990][10228:10228] CHIP:FP: Forgetting fabric 0x1
[1659508250.471029][10228:10228] CHIP:TS: Pending Last Known Good Time: 2022-08-01T09:44:35
[1659508250.471248][10228:10228] CHIP:TS: Previous Last Known Good Time: 2022-08-01T09:44:35
[1659508250.471279][10228:10228] CHIP:TS: Reverted Last Known Good Time to previous value
[1659508250.471327][10228:10228] CHIP:CTL: Shutting down the commissioner`

`Pls run this test in chip tool interactive mode  using the command  ./chip-tool interactive start 
operationalcredentials subscribe nocs  1 30 2 0 --fabric-filtered false  --keepSubscriptions 0
On TH(Chiptool)2,Verify TH  subscribes to the Node Operational Credentials cluster NOCs attribute, min interval 1, max interval 30, keep subscriptions false, fabric-filtered false and save SubscriptionId as subscription [... cut ...]
[1659598415.565587][3022:3027] CHIP:EM: Sending Standalone Ack for MessageCounter:207708624 on exchange 29451i
[1659598415.565652][3022:3027] CHIP:IN: Prepared secure message 0xffffa0dbb9e8 to 0x0000000000000002 (1)  of type 0x10 and protocolId (0, 0) on exchange 29451i with MessageCounter:125509364.
[1659598415.565690][3022:3027] CHIP:IN: Sending encrypted msg 0xffffa0dbb9e8 with MessageCounter:125509364 to 0x0000000000000002 (1) at monotonic time: 0000000000820B0E msec
[1659598415.565811][3022:3027] CHIP:EM: Flushed pending ack for MessageCounter:207708624 on exchange 29451i`

lisaaaas, 2023-01-18:
> 
Why is it not a valid instruction, is there a change in the test case, which is the result of me executing the latest test case

bzbarsky-apple, 2023-01-18:
> Why is it not a valid instruction

The only non-optional argument "chip-tool interactive" takes is "start", and the optional arguments it takes do not include `--keepSubscriptions`.  So trying to run `chip-tool interactive start operationalcredentials subscribe nocs 1 30 2 0 --fabric-filtered false --keepSubscriptions 0` will just give you an error like so:

```
[1674055141708] [59951:41441089] [TOO] InitArgs: Optional argument operationalcredentials does not exist.
Usage:
  ./out/debug/standalone/chip-tool interactive start [--paa-trust-store-path] [--cd-trust-store-path] [--commissioner-name] [--commissioner-nodeid] [--use-max-sized-certs] [--only-allow-trusted-cd-keys] [--trace_file] [--trace_log] [--trace_decode] [--ble-adapter]
```

I don't know where the instructions you are following come from, but if they say to run that command, they are wrong.

Also, none of this has to do with this initial issue, and should probably be in a new issue....

lisaaaas, 2023-01-30:
> This test step is in the test documentation

[image]

lisaaaas, 2023-01-30:
The first instruction is executed on terminal 1 and the second instruction is executed on terminal 2
[image]

bzbarsky-apple, 2023-01-30:
@lisaaaas Please file issues with the test plan verification steps in https://github.com/CHIP-Specifications/chip-test-plans/

bzbarsky-apple, 2023-01-30:
But that said, what those instructions are _trying_ to say, and not saying very clearly, is to run `chip-tool interactive start` and then at the interactive mode prompt to run `operationalcredentials subscribe nocs 1 30 2 0 --fabric-filtered false --keepSubscriptions 0`.  Whether it should work depends most immediately on whether the node was in fact commissioned with node id 2.

lisaaaas, 2023-01-30:
> But that said, what those instructions are _trying_ to say, and not saying very clearly, is to run `chip-tool interactive start` and then at the interactive mode prompt to run `operationalcredentials subscribe nocs 1 30 2 0 --fabric-filtered false --keepSubscriptions 0`. Whether it should work depends most immediately on whether the node was in fact commissioned with node id 2. And again, none of that has to do with this long-closed issue.

yeah，I am not sure why I operate on node 2 in interactive mode. I would like to know whether there are preconditions for executing test case TC-ACE-1.2

lisaaaas, 2023-01-30:
> @lisaaaas Please file issues with the test plan verification steps in https://github.com/CHIP-Specifications/chip-test-plans/

This link opens with an error

bzbarsky-apple, 2023-01-30:
> This link opens with an error

That probably means you don't have access to that repository.  Please talk to CSA about that.

lisaaaas, 2023-01-31:
> commit f98f73380c22f410ff6f323f182774a78154b1ce (HEAD -> main, origin/main) , I had run below test steps in Interactive mode. logs are attached from Step 16 to 19. [TC-ACE - 1.2 ] Issues IN STEPS 18 [...]

After looking at your test log,I am not sure why I operate on node 2 in interactive mode. I would like to know whether there are preconditions for executing test case TC-ACE-1.2