project-chip/connectedhomeip #21654: Need support in Darwin tool for discover commissionables

Garik098, 2022-08-05:
For the testcase TC-SC-4.2 step 5, we need to check it through discover commissionables attribute which was present in chip tool but not in Darwin tool. This step is a shall requirement in the Spec (Commissioners SHALL silently ignore TXT record keys that they do not recognize). Needed support in Darwin tool to execute this test case.

[image]
[image]
[image]

bzbarsky-apple, 2022-08-05:
I don't think this test plan makes sense, sorry....

This test plan is expecting a DUT to "report" things... but why would a commissioner device do that in practice?  Let's say the DUT were a commissioner device.  Where/why would it "report" the DNS-SD instance name of the thing it discovers?  No user is going to want to see that.... Similarly for reporting the TXT keys: an actual device is not going to show those anywhere.

Concretely, the Darwin framework does not have any discovery APIs apart from "here's a payload; open a PASE session to the device".  It does not have a generic scanning API, and does not report any of the discovery results to its API consumers.

The spirit of this test plan would be that an advertisement device with the extra TXT keys is treated the same as one without, and commissioned exactly the same way, I guess...

woody-apple, 2022-09-06:
Darwin Issue Review: Moving to future, as we don't believe this is required.

geoffroyletourneur, 2022-10-20:
We would like to support discover commissionables from chiptool iOS app. 
I guess this issue is also available on iOS. I mean by that there is no possibilty to discover commissionables from chiptool app. 
Let me know if I'm wrong and if there is another way.

closed 2023-04-30 by stale[bot]