project-chip/connectedhomeip #21662: [crash] [MTRBaseClusterThermostat initWithDevice:endpoint:queue:]_block_invoke

aajain-com, 2022-08-05:
#### Problem
SHA `aa9457e6b94b735076dff6297176183bf9780177 `

```
Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Subtype: KERN_INVALID_ADDRESS at 0x8020000000000030 -> 0x0000000000000030 (possible pointer authentication failure)
Exception Codes: 0x0000000000000001, 0x8020000000000030
VM Region Info: 0x30 is not in any region.  Bytes before following region: 68719476688
      REGION TYPE                 START - END      [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      commpage (reserved)     1000000000-7000000000 [384.0G] ---/--- SM=NUL  ...(unallocated)
Termination Reason: SIGNAL 11 Segmentation fault: 11
Terminating Process: exc handler [234]

Triggered by Thread:  4
```

```
Thread 4 Crashed ↩::   Dispatch queue: com.csa.matter.framework.controller.workqueue
0   Matter                        	       0x1a1bacc7c __58-[MTRBaseClusterThermostat initWithDevice:endpoint:queue:]_block_invoke + 64 (connectedhomeip/src/darwin/Framework/CHIP/zap-generated/MTRBaseClusters.mm:42062)
1   Matter                        	       0x1a1bacc6c __58-[MTRBaseClusterThermostat initWithDevice:endpoint:queue:]_block_invoke + 48 (connectedhomeip/src/darwin/Framework/CHIP/zap-generated/MTRBaseClusters.mm:42062)
2   libdispatch.dylib             	       0x1e22372d0 _dispatch_client_callout + 20 libdispatch/src/object.m:560)
3   libdispatch.dylib             	       0x1e2246c50 _dispatch_sync_invoke_and_complete_recurse + 64 (libdispatch/src/queue.c:1023)
4   libdispatch.dylib             	       0x1e22466d0 _dispatch_sync_f_slow + 172 (libdispatch/src/queue.c:1780)
5   Matter                        	       0x1a1bacbb0 -[MTRBaseClusterThermostat initWithDevice:endpoint:queue:] + 224 (connectedhomeip/src/darwin/Framework/CHIP/zap-generated/MTRBaseClusters.mm:42060)
```

```
2022-08-04 12:37:04.0850 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824080] [234:4125] CHIP: [EM] Flushed pending ack for MessageCounter:194090099 on exchange 59484i
2022-08-04 12:37:04.0937 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824089] [234:3078] CHIP: [CTL] Response Time: PFvPvsE+1989230764
2022-08-04 12:37:04.0962 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824091] [234:3078] CHIP: [DMG] SendReadRequest ReadClient[0x155e808d0]: Sending Read Request
2022-08-04 12:37:04.0964 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824092] [234:3078] CHIP: [IN] Prepared secure message 0x1558c9568 to 0x00000000983BF68E (10)  of type 0x2 and protocolId (0, 1) on exchange 59485i with MessageCounter:72284453.
2022-08-04 12:37:04.0965 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824092] [234:3078] CHIP: [IN] Sending encrypted msg 0x1558c9568 with MessageCounter:72284453 to 0x00000000983BF68E (10) at monotonic time: 000000000001E956 msec
2022-08-04 12:37:04.0969 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824092] [234:3078] CHIP: [DMG] MoveToState ReadClient[0x155e808d0]: Moving to [AwaitingIn]
2022-08-04 12:37:04.1563 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824152] [234:3078] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:194090100 on exchange 59485i
2022-08-04 12:37:04.1563 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824152] [234:3078] CHIP: [EM] Found matching exchange: 59485i, Delegate: 0x155e808d0
2022-08-04 12:37:04.1566 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824152] [234:3078] CHIP: [EM] Rxd Ack; Removing MessageCounter:72284453 from Retrans Table on exchange 59485i
2022-08-04 12:37:04.1566 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824152] [234:3078] CHIP: [EM] Removed CHIP MessageCounter:72284453 from RetransTable on exchange 59485i
2022-08-04 12:37:04.1572 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824152] [234:3078] CHIP: [EM] Sending Standalone Ack for MessageCounter:194090100 on exchange 59485i
2022-08-04 12:37:04.1573 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824152] [234:4125] CHIP: [CTL] Response Time: PFvPvsE+1989230764 0.064706 seconds
2022-08-04 12:37:04.1573 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824152] [234:3078] CHIP: [IN] Prepared secure message 0x16d4e12b8 to 0x00000000983BF68E (10)  of type 0x10 and protocolId (0, 0) on exchange 59485i with MessageCounter:72284454.
2022-08-04 12:37:04.1573 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824153] [234:3078] CHIP: [IN] Sending encrypted msg 0x16d4e12b8 with MessageCounter:72284454 to 0x00000000983BF68E (10) at monotonic time: 000000000001E993 msec
2022-08-04 12:37:04.1575 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824153] [234:3078] CHIP: [EM] Flushed pending ack for MessageCounter:194090100 on exchange 59485i
2022-08-04 12:37:04.1624 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824158] [234:4125] CHIP: [CTL] Response Time: PFvPvhE+2192660888
2022-08-04 12:37:04.1650 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824160] [234:4125] CHIP: [DMG] SendReadRequest ReadClient[0x153fdbbb0]: Sending Read Request
2022-08-04 12:37:04.1652 -0700 default 0x0101d 0x0 homed[234] (Matter): 🔵 [1659641824160] [234:4125] CHIP: [IN] Prepared secure message 0x158914e68 to 0x00000000983BF68E (10)  of type 0x2 and protocolId (0, 1) on exchange 59486i with MessageCounter:72284455.
2022-08-04 12:37:04.1653 -0700 default 0x0101d 0x0 homed[234] (Matter): 🔵 [1659641824160] [234:4125] CHIP: [IN] Sending encrypted msg 0x158914e68 with MessageCounter:72284455 to 0x00000000983BF68E (10) at monotonic time: 000000000001E99B msec
2022-08-04 12:37:04.1657 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824161] [234:4125] CHIP: [DMG] MoveToState ReadClient[0x153fdbbb0]: Moving to [AwaitingIn]
2022-08-04 12:37:04.2216 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824217] [234:3078] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:194090101 on exchange 59486i
2022-08-04 12:37:04.2216 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824217] [234:3078] CHIP: [EM] Found matching exchange: 59486i, Delegate: 0x153fdbbb0
2022-08-04 12:37:04.2217 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824217] [234:3078] CHIP: [EM] Rxd Ack; Removing MessageCounter:72284455 from Retrans Table on exchange 59486i
2022-08-04 12:37:04.2217 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824217] [234:3078] CHIP: [EM] Removed CHIP MessageCounter:72284455 from RetransTable on exchange 59486i
2022-08-04 12:37:04.2224 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824218] [234:3078] CHIP: [EM] Sending Standalone Ack for MessageCounter:194090101 on exchange 59486i
2022-08-04 12:37:04.2224 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824218] [234:4125] CHIP: [CTL] Response Time: PFvPvhE+2192660888 0.060024 seconds
2022-08-04 12:37:04.2225 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824218] [234:3078] CHIP: [IN] Prepared secure message 0x16d4e12b8 to 0x00000000983BF68E (10)  of type 0x10 and protocolId (0, 0) on exchange 59486i with MessageCounter:72284456.
2022-08-04 12:37:04.2225 -0700 default 0x00c06 0x0 homed[234] (Matter): 🔵 [1659641824218] [234:3078] CHIP: [IN] Sending encrypted msg 0x16d4e12b8 with MessageCounter:72284456 to 0x00000000983BF68E (10) at monotonic time: 000000000001E9D4 msec
2022-08-04 12:37:04.2227 -0700 default 0x00c06 0x0 homed[234] (Matter): 🟢 [1659641824218] [234:3078] CHIP: [EM] Flushed pending ack for MessageCounter:194090101 on exchange 59486i
2022-08-04 12:37:04.2262 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824221] [234:4125] CHIP: [CTL] Response Time: PFvPvhE+1665476950
2022-08-04 12:37:04.2263 -0700 default 0x0101d 0x0 homed[234] (Matter): 🟢 [1659641824221] [234:4125] CHIP: [DMG] SendReadRequest ReadClient[0x153fdbbb0]: Sending Read Request
2022-08-04 12:37:04.2265 -0700 default 0x0101d 0x0 homed[234] (Matter): 🔵 [1659641824222] [234:4125] CHIP: [IN] Prepared secure message 0x158914e68 to 0x00000000983BF68E (10)  of type 0x2 and protocolId (0, 1) on exchange 59487i with MessageCounter:72284457.
2022-08-04 12:37:04.2266 -0700 default 0x0101d 0x0 homed[234] (Matter): 🔵 [1659641824222] [234:4125] CHIP: [IN] Sending encrypted msg 0x158914e68 with MessageCounter:72284457 to 0x00000000983BF68E (10) at monotonic time: 000000000001E9D8 msec
[... cut ...]
2022-08-04 12:37:04.2394 -0700 default 0x00c06 0x3252 homed[234] (Matter): 🔵 [1659641824235] [234:3078] CHIP: [BLE] CancelConnection
2022-08-04 12:37:04.2394 -0700   error 0x00c06 0x3252 homed[234] (Matter): 🔴 [1659641824235] [234:3078] CHIP: [DL] Inet Layer shutdown
2022-08-04 12:37:04.2394 -0700   error 0x00c06 0x3252 homed[234] (Matter): 🔴 [1659641824235] [234:3078] CHIP: [DL] BLE shutdown
2022-08-04 12:37:04.2394 -0700   error 0x00c06 0x3252 homed[234] (Matter): 🔴 [1659641824235] [234:3078] CHIP: [DL] System Layer shutdown
```

Apple reference: rdar://98142314

woody-apple, 2022-08-30:
This is no longer showing up in crashes.

closed 2022-08-30 by woody-apple

bzbarsky-apple, 2022-08-31:
This was fixed by https://github.com/project-chip/connectedhomeip/pull/21504