project-chip/connectedhomeip #21664: chip-tool scans for BLE devices when it should not

bluebin14, 2022-08-05:
#### Problem
After opening a commisioning window chip-tool is used as a new commissioner to scan for the device. As per spec this can only be done on IP network, yet chip-tool scans primarily for BLE wasting a great amount of bandwith and log output, interfering with the purpose of the command. The command used is "chip-tool pairing code".

 #### Proposed Solution
Fix "chip-tool pairing code" so that it does not scan for BLE unless Thread credentials are provided. The reason: see above, plus without network credentials the commissioning would fail anyway.

bzbarsky-apple, 2022-08-05:
@vivien-apple Any objections to changing `pairing code` to munge the onboarding payload to just set the "on-network" flag in its discovery bitmap?

bluebin14, 2022-08-05:
Note that the QR code shipping with Thread devices will have the BLE flag set. The same QR code should be still usable however when opening basic commissioning window, which obviously needs IP. Since at that point the device is already on IP, obeying the BLE flag would cause definite failure. The only way to reuse the QR code: if no network credentials provided then ignore BLE flag, even if it is set.

bzbarsky-apple, 2022-08-05:
@bluebin14 SetUpCodePairer always does on-network discovery, no matter what, because the spec says to do that.  So "obeying the BLE flag" is not a problem.

Per Slack discussion, by the way, this issue has nothing to do with "pairing code".  This is about the behavior of the test `PairWithCode` command in YAML.

bluebin14, 2022-08-05:
Having a yaml fix is good as long as chip-tool itself is usable in a mode that skips BLE scanning.

woody-apple, 2022-08-16:
Cert Blocker Review: @cjandhyala to figure out if this is blocking test cases.

cjandhyala, 2022-08-19:
per conversation with John, the issue happens when running with a bunch of devices doing BLE advertising in a lab environment, the pairing with code command is spending too much time in BLE scanning, when it is not necessary to do BLE scanning because when the DUT is in commissioning mode after opening window with ECM/BCM, chip-tool should scan only on IP transport and not required to scan BLE adv. This is not an issue if they have 1-2 devices around. 

The suggested fix was, to fix pairing with code cmd in the chip-tool not scan for ble unless the user provides BLE/thread credentials.

PR #22104 (2022-08-25): [SetUpCodePairer] Only search on network if no credentials are provid…
…ed to chip-tool pairing code commands

#### Problem

In #21664 `chip-tool` is used to add an other commissioner after having opened a commissioning window. In theory this operation should only happens on-network. But the fact that the qr code provided for the initial pairing is used get the setup code pairer to issue a BLE scan. That is not necessary and that is also an issue when testing in a lab with a lot of ble devices.

fix #21664

#### Change overview
 * If no credentials are provided to the setup code pairer issue a [... cut]
PR #22166 (2022-08-29): [SetUpCodePairer] Only search on network if no credentials are provid…
…ed to chip-tool pairing code commands

#### Problem

In #21664 `chip-tool` is used to add an other commissioner after having opened a commissioning window. In theory this operation should only happens on-network. But the fact that the qr code provided for the initial pairing is used get the setup code pairer to issue a BLE scan. That is not necessary and that is also an issue when testing in a lab with a lot of ble devices.

fix #21664

#### Change overview
 * If no credentials are provided to the setup code pairer issue a [... cut]
closed 2022-08-25 by woody-apple