project-chip/connectedhomeip #21681: chip-tool: add more flexibility in PAA root folder selection

bluebin14, 2022-08-05:
#### Problem
chip-tool currently can only select a PAA root folder by command line argument. This presents an integration problem in various existing certification tools since a large amount of calls in several scripts would need to be changed. Instead there is a need for more flexibility in specifying PAA root folder which only requires a backwards compatible change to chip-tool.

 #### Proposed Solution
The following to be applied to chip-tool if --paa-trust-store-path is not given:

    if ~/paa-root-certs exists use that
    else if environment variable PAA_ROOT_CERTS exists use that

tcarmelveilleux, 2022-08-05:
Assigned to @kpschoedel, he may have time within the next few days. Otherwise we will reassign.

I would recommend against `~/paa-root-certs` -> this can be achieved by having `PAA_ROOT_CERTS` environment variable set to `~/paa-root-certs` in the shell's profile by default, which would avoid special casing.

bluebin14, 2022-08-05:
yes, PAA_ROOT_CERTS should be sufficient

PR #21691 (2022-08-08): chip-tool: add more flexibility in PAA root folder selection
#### Problem

Fixes #21681 chip-tool: add more flexibility in PAA root folder selection

#### Change overview

If `--paa-trust-store-path` is not given, and the environment variable
`CHIPTOOL_PAA_TRUST_STORE_PATH` exists, use its value.

#### Testing

Manual confirmation of the logged `mPaaTrustStorePath.Value()`.
commit 34a8570d (2022-08-08): chip-tool: add more flexibility in PAA root folder selection (#21691)

* chip-tool: add more flexibility in PAA root folder selection

#### Problem

Fixes #21681 chip-tool: add more flexibility in PAA root folder selection

#### Change overview

If `--paa-trust-store-path` is not given, and the environment variable
`CHIPTOOL_PAA_TRUST_STORE_PATH` exists, use its value.

#### Testing

Manual confirmation of the logged `mPaaTrustStorePath.Value()`.

* restyled
closed 2022-08-08 by tcarmelveilleux