project-chip/connectedhomeip #21700: [TC-SWTCH-2.2] -  MultiPressComplete event is not subscribed

somu1710, 2022-08-06:
**Description:**
Currently subscription to event is happened for the  initial-press,  short-release and multi-press-ongoing events.But MultiPressComplete event  is not actually subscribed.While using echo command  to triger the  MultiPressComplete event  has triggered successfully.But  sending the subscribe event MultiPressComplete command it shows the response alone not the actual event  triggered.

**Actual outcome:**
[1659702279.069789][4374:4379] CHIP:DMG: ReportDataMessage =
[1659702279.069813][4374:4379] CHIP:DMG: {
[1659702279.069834][4374:4379] CHIP:DMG: 	SubscriptionId = 0x815ab601,
[1659702279.069856][4374:4379] CHIP:DMG: 	InteractionModelRevision = 1
[1659702279.069876][4374:4379] CHIP:DMG: }
[1659702279.069911][4374:4379] CHIP:DMG: MoveToState ReadClient[0xffff9c00afc0]: 

**Expected outcome:**
[1659695330.923264][3912:3918] CHIP:TOO:          Event number: 2
[1659695330.923283][3912:3918] CHIP:TOO:          Priority: Info
[1659695330.923302][3912:3918] CHIP:TOO:          Timestamp: 17754336
[1659695330.923326][3912:3918] CHIP:TOO:   **MultiPressComplete:** {
[1659695330.923346][3912:3918] CHIP:TOO:     **PrevioiusPosition: 1**
[1659695330.923366][3912:3918] CHIP:TOO:     **TotalNumberOfPressesCounted: 1**

**Commands used:**
To Trigger:

`echo '{"Name":"MultiPressComplete","PreviousPosition":1,"TotalNumberOfPressesCounted":1}' > /tmp/chip_all_clusters_fifo-<PID>  (PID of DUT)`

**To subscribe:**
```
./chip-tool interactive start
switch subscribe-event multi-press-complete 1 100 1 1 
```

```

DUT - Raspi
App - All-clusters-app
Platform - Linux
```
**Test plan Link:**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/switch.adoc#322-tc-swtch-22-primary-functionality-with-server-as-dut

**Readme link:**
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/linux#trigger-switch-events

**Logs for reference:**
[TC-SWTCH-2.2 DUT.txt](https://github.com/project-chip/connectedhomeip/files/9274765/TC-SWTCH-2.2.DUT.txt)
[TC-SWTCH-2.2 EVENTS.txt](https://github.com/project-chip/connectedhomeip/files/9274768/TC-SWTCH-2.2.EVENTS.txt)
[TC-SWTCH-2.2 CTRL.txt](https://github.com/project-chip/connectedhomeip/files/9274766/TC-SWTCH-2.2.CTRL.txt)

somu1710, 2022-08-06:
@yufengwangca Could you please take a look at this issue?

bzbarsky-apple, 2022-08-07:
https://github.com/project-chip/connectedhomeip/pull/21702 fixes.

closed 2022-08-08 by jmartinez-silabs