project-chip/connectedhomeip #21703: When Android APP reestablised connection with device by calling "updateDevice" or "Connect", it would crash.

Hugh-Bao, 2022-08-08:
#### Problem
The code was based on SVE, the Matter Android app and the Matter device that had established a connection. During our test, the Matter Android app would be repeatedly killed and then restarted. When the Matter app reestablished a connection with the device ("updateDevice"  or "Connect"  was called), the app would crash at location of "lib/IntrusiveList.h" : "Prepend" : VerifyOrDie(!node->IsInList()) ; the function called hierarchy is UpdateDevice > proxy->LookupPeerAddress > Resolver::Instance().LookupNode(request, mAddressLookupHandle) > mActiveLookups.PushBack(&handle) > PushBack > Prepend > VerifyOrDie(!node->IsInList())

bzbarsky-apple, 2022-08-08:
@Hugh-Bao Complete logs from the Android app (and stacks to the assertion failure) would be useful.

Hugh-Bao, 2022-08-09:
@bzbarsky-apple  It seems like we aren't calling the relative Matter APIs  in a thread-safe manner. We added the mutex lock in our  relative location of code, then this problem was resolved. I will close this issue, thank you Boris.

closed 2022-08-09 by Hugh-Bao