project-chip/connectedhomeip #21704: Startup event should be the first event recorded

pankore, 2022-08-08:
#### Problem
According to specs, the Startup event should be the first Data Model event recorded by the node after the reboot process.
However, the Boot-reason event is the first event recorded instead. 
See attached Server initialisation log.

 #### Proposed Solution
Emit Startup event earlier

[server_init.log](https://github.com/project-chip/connectedhomeip/files/9278575/server_init.log)

bzbarsky-apple, 2022-08-08:
There's another problem too: we're emitting BootReason events for Matter stack shutdown+startup even if there was no reboot.  Maybe that's OK, but it's not clear whether we should do that.

Anyway, right now we emit the StartUp event at the end of the boot process, but emitting BootReason from inside the cluster init code, which comes before we have finished boot.  We should probably fix that second one just because we might be emitting it when we're not in a fully initialized state.

That said, what we are doing right now is not technically violating the spec (since StartUp being first is a SHOULD, not a SHALL)...

mrjerryjohns, 2022-08-10:
Not a high priority honestly.

closed 2023-10-15 by stale[bot]