project-chip/connectedhomeip #21716: Allow IsCurrentInterfaceUsable platform override for MinimalMdns

s-mcclain, 2022-08-08:
#### Problem
MinimalMdnsServer.h defines it's own function, `IsCurrentInterfaceUsable()` for indicating which interfaces on the device should be used for broadcasting mdns queries/responses.
This function simply returns true unless it contains `lo`. I'm sure there's several platforms which contain interfaces outside of `lo` which we would like to have disabled, but there's currently no way to configure which platforms to enable/disable.

 #### Proposed Solution
Allow for platforms to be able to either:
* Override the function `IsCurrentInterfaceUsable()` OR
* Allow for platform to be able to specific which platforms they wish to have enabled.

bzbarsky-apple, 2022-08-08:
A related question: should the same deny-list apply to broadcasting group messages?  @andy31415

andy31415, 2023-08-09:
We have some support for this by defining how interfaces are iterated upon in `AddressPolicy.h` where listen endpoints and valid ip addresses for endpoints are defined. We have a libnl implementation that is slightly better than the default (it will ignore temporary IPs or interfaces unable to broadcast).

We could expand on that support to allow platforms to override additional things.

Could we have a list of useful things to override? I think:
   - what we listen on (seems already done?)
   - where to broadcast queries (need explanation on why this is different from where to listen on)
   - ip address priority sorting (? just a guess .. we had several changes on ip prioritization ... this may not be minmdns specific and is more on the client than server side)
   - other?