project-chip/connectedhomeip #21724: chip-tool subscriptions shutdown-subscriptions should not require clients to know all fabric indexes

bluebin14, 2022-08-08:
#### Problem
chip-tool subscriptions shutdown-subscriptions currently requires clients to know all fabric indexes and in the typical use case it requires them to call the tool for all fabrics separately. This makes it impractical to use, especially in scripts that do not require advance factory reset and hardcoded fabric index of 1. 

 #### Proposed Solution
Make it practical to use by removing fabricIndex argument and clearing all subscriptions. Those who wish individual control over subscriptions can still use the singular shutdown-subscription option as before.

bzbarsky-apple, 2022-08-08:
So it seems like the operations that people might want are:

1. Shut down a single subscription (this is currently `shutdown-subscription`)
2. Shut down all subscriptions to a given node (this is currently `shutdown-subscriptions`, but requires specifying a fabric-index, which it could instead derive from the identity for the command).
3. Shut down all subscriptions to all nodes (this currently does not exist).

Note that "a given node" is necessarily tied to a given fabric, so if there are subscriptions on multiple fabrics there's no nice way to clear them all short of shutting down all subscriptions period.

@bluebin14 Does that match your understanding?  It sounds like we should add a `shutdown-all-subscriptions` command, and probably change `shutdown-subscriptions` to use identity instead of an explicit fabric-index.

bluebin14, 2022-08-08:
Yes, anything would work as long as there is a way to do full clear without having to know all fabric indexes.

closed 2022-08-10 by woody-apple